ChristianTremblay/pyhaystack #1: isConnected variable doesn't keep track of connectivity to server

ChristianTremblay, 2016-01-11:
If session has been defined for too long, it will be closed by server. A call of session.something will fail.
Maybe we should reconnect more often, or deal with that differently...

sjlongland, 2016-01-12:
In the past, I've usually taken the current timestamp (from `time.time()`), added some expiry time to it, and stored that.

`isConnected` then simply becomes a property:

```
@property
def isConnected(self):
    return self._auth_expiry > time.time()
```

If the implementation provides an expiry (e.g. Widesky does), then we just use that expiry time.  Otherwise, we can add some nominal expiry time (user configurable) to the current time.

From what I gather with Niagara AX, it uses HTTP basic authentication, so doesn't establish a true "session", the credentials are provided on *every* request.  HTTP digest authentication would be the same.

ChristianTremblay, 2016-04-09:
The async engine will handle that

closed 2016-04-09 by ChristianTremblay