ChristianTremblay/pyhaystack #3: add Exceptions instead of print

ChristianTremblay, 2016-01-16:


ChristianTremblay, 2016-01-17:
in this issue I also propose syntax improvement for histories

sjlongland, 2016-01-18:
On the histories front, how's this for size?

```
In [1]: import logging; logging.basicConfig(level=logging.DEBUG); logging.getLogger('iso8601').setLevel(logging.INFO)
In [2]: from pyhaystack.client import WideskyClient; wsc=WideskyClient.Connect('http://localhost:8080/',username='…',password='…')
INFO:requests.packages.urllib3.connectionpool:Starting new HTTP connection (1): localhost
DEBUG:requests.packages.urllib3.connectionpool:"POST /login/ HTTP/1.1" 200 472
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/about, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/about HTTP/1.1" 200 None
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/read?filter=site, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/read?filter=site HTTP/1.1" 200 None
DEBUG:pyhaystack.client.authenticate:Connected to haystack instance on 61e8827a42d6 (protocol 2.0, implementation Widesky Semantic Database Toolkit 0.0.1).  Time Zone used : Australia/Brisbane
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/read?filter=his, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/read?filter=his HTTP/1.1" 200 None

In [3]: p=wsc['my.site.submeter.1.energy']
DEBUG:pyhaystack.client.getitem:Retrieving single point my.site.submeter.1.energy
DEBUG:pyhaystack.client.getitem:Not yet retrieved point my.site.submeter.1.energy
DEBUG:pyhaystack.client.getitem:Need to retrieve points ['my.site.submeter.1.energy']
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/read?id=%40my.site.submeter.1.energy, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/read?id=%40my.site.submeter.1.energy HTTP/1.1" 200 None
DEBUG:pyhaystack.client.getitem:Retrieved [u'my.site.submeter.1.energy']

In [4]: import datetime

In [5]: p.his_write([{'ts': datetime.datetime.now().replace(microsecond=0), 'val': 98765.0}])DEBUG:pyhaystack.client.http:Submitting POST request for http://localhost:8080/api/hisWrite, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Content-Type': 'text/zinc'}, data: u'ver:"2.0" id:@my.site.submeter.1.energy\nts,val\n2016-01-18T16:56:27+10:00 Brisbane,98765.0\n'
DEBUG:requests.packages.urllib3.connectionpool:"POST /api/hisWrite HTTP/1.1" 200 None

In [6]: r=p.his_read(rng='today')
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/hisRead?range=today&id=%40my.site.submeter.1.energy, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDQ5NzQ3NjMsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.rBGurS6MEH8XhgFaM3N5vwKsbDELL36-6fsvm6O2nLE', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/hisRead?range=today&id=%40my.site.submeter.1.energy HTTP/1.1" 200 None
DEBUG:pyhaystack.client.hisRecord.my.site.submeter.1.energy:Received result set: <hszinc.grid.Grid object at 0x7f7dcba676d0>

In [7]: r.data
Out[7]: 
[energy]
2016-01-15 09:47:46+10:00      123.450
2016-01-15 09:48:45+10:00      123.450
2016-01-15 09:48:56+10:00     6789.010
2016-01-15 10:39:50+10:00     6789.010
2016-01-18 10:47:12+10:00      123.456
2016-01-18 10:47:15+10:00      123.456
2016-01-18 10:47:16+10:00      123.456
2016-01-18 16:56:27+10:00    98765.000
dtype: float64
```

There are three ways to get an instance of a `HaystackPoint`:
* specifying a filter using `find_point`: this returns a `dict` of matching points, point ID is the key
[... cut ...]
* specifying a single point ID to `__getitem__`: returns that point or `None` if not found.

I'm tossing up whether `KeyError` should be thrown in the "single point not found" case.

As for reading historical data, `HaystackPoint` has a `__getitem__` method that can take:
* a single value, which is passed to `his_read` as the `rng` parameter
* a `slice` object (created by calling `point[start:end]`, the `slice`'s `start` and `stop` get passed to `his_read` as `start` and `end`.

Detail is in commit 118d4f097f7a3dd7ec6898fc20a0ab90af8764bb

sjlongland, 2016-01-18:
And speaking of `find_point`, I came up with a bit of a hacky way to build up a filter string that abuses Python's operators which gives us hopefully a more readable form:

https://github.com/vrtsystems/pyhaystack/commit/5ca34bf6be0d9f2472e3303ddf25dfe2a32f156b

So one can do something like this:
```
In [1]: from pyhaystack.util import filterbuilder as fb

In [2]: str(fb.Field('his') & \
                ( fb.Field('tz') == fb.Scalar('Brisbane') ))
Out[2]: 'his and ( tz == "Brisbane" )'

In [3]: from pyhaystack.client import WideskyClient; wsc=WideskyClient.Connect('http://localhost:8080/',username='…',password='…')

In [4]: import logging; logging.basicConfig(level=logging.DEBUG); logging.getLogger('iso8601').setLevel(logging.INFO)
In [5]: p=wsc.find_points(str(fb.Field('his') & \                ( fb.Field('tz') == fb.Scalar('Brisbane') )))
DEBUG:pyhaystack.client.http:Submitting application/json GET request for http://localhost:8080/api/read?filter=his+and+%28+tz+%3D%3D+%22Brisbane%22+%29, headers: {'X-Access-Token': u'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE0NTM3MDgyMzQ1OTEsInVzZXIiOnsibmFtZSI6InN1cGVydXNlciIsInJvbGUiOiJhZG1pbiIsInVzZXJuYW1lIjoic3VwZXJ1c2VyIiwicGFzc3dvcmQiOiIkMmEkMTAkcFVoM29CbkRNMjZnMU0xb1Audnd1ZTBEWkI5bmo5Snd4VWlkelBILjVQbmhnaWxwVEpkWUcifX0.eBbIT4pbauj7EiuAQQbbXCUqApv5MZkN8gHfFBteIq4', 'Accept': 'application/json'}
DEBUG:requests.packages.urllib3.connectionpool:"GET /api/read?filter=his+and+%28+tz+%3D%3D+%22Brisbane%22+%29 HTTP/1.1" 200 None

In [6]: p
Out[6]: 
{u'my.site.main.meter.energy': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.main.meter.energy'),
 u'my.site.main.meter.power': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.main.meter.power'),
 u'my.site.submeter.1.energy': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.submeter.1.energy'),
 u'my.site.submeter.1.power': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.submeter.1.power'),
 u'my.site.submeter.2.energy': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.submeter.2.energy'),
 u'my.site.submeter.2.power': HaystackPoint(<pyhaystack.client.WideskyClient.Connect instance at 0x7f703d7fd128>, u'my.site.submeter.2.power')}
```

ChristianTremblay, 2016-04-09:
Closing this as it's been mostly implemented

closed 2016-04-09 by ChristianTremblay