ChristianTremblay/pyhaystack #4: Needle and Pyhaystack

nehalecky, 2016-01-20:
Hi Pyhastack team.

So glad to see this effort, connecting SkySpark and Niagara frameworks to harness the significant utility provided by the Python scientific stack is something needed for this building energy community. :)

Myself and a few others in the open source community (@christopher-hartley) wanted to understand better how pyhaystack compares with the [Smart-Green/needle](https://github.com/Smart-Green/needle) library? Browsing across the commit history shows that it is a fork of pyhaystack, and indeed it seems both projects goals are totally aligned.  

Any feedback would be appreciated—thanks so much!

sjlongland, 2016-01-20:
Well, from my perspective, I had no idea that the `needle` project existed, having done a search on pypi and Google to try and locate such a framework.  I saw reference to `pyhaystack` on the Project Haystack forums and got in touch with @ChristianTremblay to see what the current status was.

My work so far has been to get a decent ZINC parser/dumper, since that seems to be the lowest common denominator format used by Project Haystack implementations.  The result of that was https://github.com/vrtsystems/hszinc which is a BSD-licensed clean-room implementation of the ZINC format (using https://github.com/erikrose/parsimonious to do the heavy lifting).

Having done this, I'm looking to get pyhaystack into some sort of shape so it can be used to push data into and pull data out of a Project Haystack system.

sjlongland, 2016-01-20:
For what it's worth, just had a stab at merging the two trees: https://github.com/vrtsystems/pyhaystack/tree/needle

nehalecky, 2016-01-20:
@sjlongland, kudos. I don't see any (updated) status of build? Did it pass the test suite?

As I understand that [pyhastack](https://github.com/ChristianTremblay/pyhaystack) is the original repo (and needle a fork), I'm going to ask @sudo-Whateverman to jump onto this thread to continue discussions on merging these two efforts. 

Reposting his thoughts, here:
> Although we are using a fork of pyhaystack, we have made a few changes to the original code which make a merge problematic. While the author has originally used python2 as for the code, he has since switched to python3. Also, some changes were introduced to make the code work for skyspark. While those changes have made it possible to connect the client to skyspark, we haven't tested it for Niagara.

This seems within the realm of possibility. :) Let's make this happen to better the data analysis tools available to Niagara / SkySpark / Haystack users and industry at large! :clap:

sudo-Whateverman, 2016-01-20:
A little bit of history about our fork:
we found <a href='https://bitbucket.org/christiantremblay/pyhaystack/overview'> pyhayastack on  bitbucket  </a> and forked the project that was seemingly abandoned, after that we continued developing for python2 and skyspark/haystack. 
I think the python <a href='https://pypi.python.org/pypi/future'> future project </a> project can be used to maintain a codebase that we could use both on python 2 & 3.

sjlongland, 2016-01-20:
I've been using `six` for a lot of my cross-Python integration work, but I'll admit I'm very new to the Python 3 way of doing things, so still have lots to learn.

sudo-Whateverman, 2016-01-20:
BTW, as a sidenote: 
please ignore the MangoClient.py file, it is a work in progress for mango automation api that is a bit raw. 
It has no connection to the skyspark platform.

sjlongland, 2016-01-20:
Just doing some further merge work, one of the big merge blockers was my whitespace clean-up, removing trailing spaces and converting files to Unix line ending format.  (Seeing ^M in diffs looks ugly!)

ChristianTremblay, 2016-01-21:
That is the thing I like about Open Source, there's always some surprises :) 

When I first tried implementing pyhaystack (this was my first Python project) I was really targetting histories at first and using Pandas... and the more I was reading about Python, the more I thought I should head with Python 3 (yes I'm a naive man). I should not have waited for approbation I pushed the project to Pypi... I would have had more return from the community. The library has been private on bitbucket for too long. 

So here we are and I'm glad to know that I'm not the only one that believe that Python is probably one of the better language to use with haystack (knowing the strong of Python for data analysis and code simplicity).

I must say it would be very appreciated to be part of a more official project, with the help of you (and others) so the library become useful and work properly. 

I now that maybe some of your projects contains features, some details that should not be part of pyhaystack, but if we could merge all our related work, we could then count on a reliable pyhaystack library, supported, tested, documented, available as an import on your different project.

@sjlongland made a great work implementing hszinc, which is now tied to pyhaystack in a really close way. And it's really important for me to promote the contributors. 

It would be great to learn with you guys. Really.

PR #5 (2016-01-22): Needle
This will be the basis of the future work on pyhaystack.

This merge work done by @sjlongland , @sudo-Whateverman and @ChristianTremblay

First merge in the staged branche that will be an intermidiate step before releasing to master

This closes #4
closed 2016-04-09 by ChristianTremblay