ChristianTremblay/pyhaystack #6: Proposition to clean HaystackClient class with the use of Mixin

ChristianTremblay, 2016-01-24:
I'll commit a branch for comments

ChristianTremblay, 2016-01-24:
I know I should not do that but I've included another piece in this commit. In the request mixin, I've included a retry decorator to handle a lost of connection.
The idea is to look at the headers or the received request. If we see text/html in it, we are probably back to the login page. So we can re-authenticate and retry the function.

when re-authenticating, I don't want the histories to be read again so I added a condition to the function.

ChristianTremblay, 2016-01-24:
See 47bd727 for details

sjlongland, 2016-01-24:
Shouldn't we get a "401 Unauthorized" status code from the HTTP server if the session has expired?

ChristianTremblay, 2016-01-24:
It's probably server related. With nHaystack, if not logged, every requests end up being the login web page. This said, it would be easy to cover also the case where there's a specific 401 error...
Niagara AX seems to be a special case in many ways ! :)

ChristianTremblay, 2016-01-24:
Here's an example of the request when trying `sensors = session.find_points('sensor')` after having disconnected the session using `session.disconnect()`

```
header : {'cache-control': 'no-cache, no-store', 'content-type': 'text/html; charset=UTF-8', 'expires': 'Thu, 01 Jan 1970 00:00:00 GMT', 'date': 'Sun, 24 Jan 2016 16:09:38 GMT', 'server': 'Niagara Web Server/3.8.38.3', 'transfer-encoding': 'chunked', 'content-encoding': 'gzip'}
request : <Response [200]>
Session disconnected, Retrying in 1 seconds...
pyhaystack 0.71.0_issue6 | Authentication to http://10.137.161.52/login/
Initiating connection
Initiating authentication
User logged in...
header : {'cache-control': 'private, must-revalidate', 'content-type': 'text/zinc; charset=utf-8', 'connection': 'close', 'expires': 'Thu, 01 Jan 1970 00:00:00 GMT', 'date': 'Sun, 24 Jan 2016 16:09:41 GMT', 'server': 'Niagara Web Server/3.8.38.3'}
request : <Response [200]>
header : {'cache-control': 'private, must-revalidate', 'content-type': 'text/zinc; charset=utf-8', 'connection': 'close', 'expires': 'Thu, 01 Jan 1970 00:00:00 GMT', 'date': 'Sun, 24 Jan 2016 16:09:41 GMT', 'server': 'Niagara Web Server/3.8.38.3'}
request : <Response [200]>
Time Zone used : America/Montreal
Connection succeed with haystack on SERVISYS_FXServer (3.8.38.1) running haystack version 2.0
header : {'cache-control': 'private, must-revalidate', 'content-type': 'text/zinc; charset=utf-8', 'connection': 'close', 'expires': 'Thu, 01 Jan 1970 00:00:00 GMT', 'date': 'Sun, 24 Jan 2016 16:09:41 GMT', 'server': 'Niagara Web Server/3.8.38.3'}
request : <Response [200]>
```

ChristianTremblay, 2016-01-24:
In `def _get_request(self, url, accept=None, headers=None, **kwargs):`, I've simply added :  
``` 
[...]
        req = self.s.get(url, **kwargs)
        req.raise_for_status()
        # Detect if disconnected
        print('header : %s' % req.headers)
        print('request : %s' % req)
[...]
```

sjlongland, 2016-01-24:
Ick, so it works that way more by "accident" on the part of `nHaystack`.  Lovely. :-|

Guess we have no choice but to kludge around their kludge.

ChristianTremblay, 2016-01-25:
I think it's more Tridium related. Their web service works this way.

ChristianTremblay, 2016-04-09:
Mostly implemented and will be revised in the future

closed 2016-04-09 by ChristianTremblay