ChristianTremblay/pyhaystack #14: Failing to log in to SkySpark with correct credentials

candronikos, 2016-05-31:
Tried testing with some simple code with skyspark with the code at the bottom of this issue. The result is posted below:

# RESULT
```
False
<GetGridOperation failed>
```

# CODE
```python
from pyhaystack.client.skyspark import SkysparkHaystackSession
session = SkysparkHaystackSession(uri='http://url_to_server',
                                username='myusername',
                                password='secretpassword',
                                project = 'project')

if(__name__ == '__main__'):
    about = session.about()
    print(session.is_logged_in)
    print(about)
```

ChristianTremblay, 2016-06-01:
Thanks for reaching us. 

I've tried your snippet using niagara client and it works. Unfortunately, I don't have actually an access to a Skyspark server. 
Would you accept to share your login info in private ? I would then be able to look at it.
My email is in my Github profile

candronikos, 2016-06-01:
Hi,

I might be able to create an account on the demo project and hand you the details. I'll see if I can get permission to do this.

candronikos, 2016-06-02:
Hi Christian,

I have just created a temp account and sent you the details

ChristianTremblay, 2016-06-02:
Actively working on

sjlongland, 2016-06-02:
Is there any publicly visible documentation for SkySpark?  I'll admit I ported the old `needle` sources to the new API blindly, I don't have access to a SkySpark server to test on.

I recall looking on their website for details but didn't see any.

candronikos, 2016-06-02:
[beyon-d.net](https://www.beyon-d.net/doc/index.html) seems to be up to date

sjlongland, 2016-06-02:
Ahh, that's why I didn't spot it, completely different site. :-)

https://www.beyon-d.net/doc/docSkySpark/Rest.html#auth would have been a big help.

ChristianTremblay, 2016-06-03:
@candronikos Please have a look to : https://github.com/ChristianTremblay/pyhaystack/tree/skyspark_login

I'm able to login, retrieve site, about... but I have some problems with find_entity. Seems to be related to hszinc... I'll give credentials to @sjlongland so he can make tests.

ChristianTremblay, 2016-06-03:
Forgot one detail : it works on Python 3.... we'll have a look to make it work with Python 2

candronikos, 2016-06-03:
Tested the code on my machine. Looks good so far! I removed the details of the response body for privacy but they're correct.

```
<GetGridOperation done: <Grid>
	Columns:
		productUri
		whoami
		serverBootTime
		projName
		tz
		moduleName
		serverName
		productName
		haystackVersion
		productVersion
		moduleVersion
		serverTime
		hostModel
	Row    0: ""DETAILS_HIDDEN""
</Grid>>
```

ChristianTremblay, 2016-06-03:
Troubles appears with : 
`sensors = session.find_entity(filter_expr='sensor').result`

You can join gitter to share more details :
https://gitter.im/ChristianTremblay/pyhaystack

sjlongland, 2016-06-03:
`hszinc` is tripping up on the grid that gets returned by SkySpark.  A simple `read` of `site` entities worked, but not `equip`s, there's something there that the parser doesn't like.

I've grabbed a copy of the offending grid and I'll have a look when I get a chance.  In the meantime, you can force `pyhaystack` to use JSON serialisation by using the following:

```
import hszinc
from pyhaystack.client.skyspark import SkysparkHaystackSession
session = SkysparkHaystackSession(uri='http://xxx',
                         username='xxx, password='xxx, project='xxx', grid_format=hszinc.MODE_JSON)
```

This is not ideal, but it might get you going.

ChristianTremblay, 2016-06-07:
Remaining bug : Make Python 2 & 3 work
string encoding is giving me a hard time.

ChristianTremblay, 2016-06-08:
https://github.com/ChristianTremblay/pyhaystack/commit/578b2691638fa6ac1da20ce803dc3b7b951e8907

Solve the Python 2/3 compatibility

closed 2016-06-08 by ChristianTremblay

jjproe, 2016-10-28:
Hello. I need a Python 2.6 to Skyspark interface. I try long to make it runable. But i have the same Problem with  <GetGridOperation> failed

ChristianTremblay, 2016-10-30:
I can't guaranty it'll work on 2.6 as we use a lot of features that are 2.7-3.
Can you be more precise about the bugs you meet ?
Also, be sure to make your tests with dev branch as some bugs have been noticed in Skyspark with master branch.

sjlongland, 2016-10-30:
I would seriously not bother with Python 2.6.  Most platforms have moved on from that release, the only OSes I can think of that depend on that release are circa 2010 or earlier, which should be considered unsupported security liabilities.

Python 2.7 is still supported because Python 3.x support was in its infancy until very recently on most Linux distributions (IMO it's a mess in Debian even in Jessie), and really is on borrowed time: the upstream Python project wants to kill off the 2.x branch.

If you're starting from scratch, I'd seriously recommend going to Python 3.4 or later if possible, as this was the first of the 3.x series that was particularly well supported.  If backward compatibility with Python 2 modules is needed, then recent 2.7 is the best option.

jjproe, 2016-10-31:
Thanks, but the system i will connect runs under SLES 11 (python 2.6.5).
I had read something about "python" environments.. and the possibility to
run both Versions..
But the effort is too big for me...

I only will get a points-list from SkysPark and perhaps import points from
foreign system..

My idea is to minimize your pyhaystack (only communication to SyksPark).. I
don't need numpy or other numerical manipulations under python..
And if hszinc make trouble to run under this python Version, i'll change to
json...

Are my reflections OK ?

sjlongland, 2016-10-31:
On 31/10/16 20:50, jjproe wrote:
> And if hszinc make trouble to run under this python Version, i'll change to
> json...

Won't make a sceric of difference.  `hszinc` is used in both modes, just
in ZINC mode it does its own serialisation/deserialisation, whereas in
JSON mode, the `json` module does the heavy lifting.

The same classes are used to represent values in both modes.

There's talk of a Python 2.7 binary as part of the OpenSuSE distribution:
https://forums.suse.com/showthread.php?5293-Python-2-7-for-SLES-11-SP3

SLES 11 is quite dated by the looks of things, so you likely won't see
anything official in terms of Python 2.7 or 3.x.  SLES 12 might be worth
investigating, the other option might be to look at Docker and running a
newer Python environment in a container.
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

ChristianTremblay, 2016-11-01:
Locking the issue and linking here : https://github.com/ChristianTremblay/pyhaystack/issues/31