ChristianTremblay/pyhaystack #15: Skyspark : history range

ChristianTremblay, 2016-06-07:
Actually, it seems that Skyspark REST api need the quotes when sending range in a hisRead command : 

```
sensors_his = session.his_read_series('MyPointId.ReturnTemp', rng='"2016-06-05,2016-06-06"').result
```

This makes use of slice impossible : 
```
import datetime
rng=slice(datetime.date(2016,6,4),datetime.date(2016,6,5))
sensors_his = session.his_read_series('MyPointId.ReturnTemp', rng=rng).result
```

Gives : 
```
HaystackError: sys::CastErr: java.lang.ClassCastException: fan.sys.Date cannot be cast to java.lang.String
```

We'll submit this in www.project-haystack.org and see what's going on.
Actually, only Skyspark is affected by this.

ChristianTremblay, 2016-06-07:
Refer to : http://project-haystack.org/forum/topic/401

candronikos, 2016-06-07:
Possibly related issue below:

```python
print("With descriptive string:")
print(session['project'].his_read_series(points.result[0]['id'], rng="yesterday"))

from datetime import date
dates = slice(date(2016,5,1),date(2016,5,31))
print("With date slice:")
print(session['project'].his_read_series(points.result[0]['id'], rng=dates))
```

gives:

```
PS C:\dir\to\code> python .\source.py
With descriptive string:
<HisReadSeriesOperation done: 2016-06-06 00:00:02+10:00    21.190001
2016-06-06 00:15:03+10:00    21.190001
2016-06-06 00:30:03+10:00    21.190001
2016-06-06 00:45:03+10:00    21.190001
2016-06-06 01:00:03+10:00    21.190001
2016-06-06 01:15:04+10:00    21.190001
2016-06-06 01:30:02+10:00    21.190001
2016-06-06 01:45:07+10:00    21.190001
2016-06-06 02:00:08+10:00    21.190001
2016-06-06 02:15:03+10:00    21.190001
2016-06-06 02:30:03+10:00    21.190001
2016-06-06 02:45:03+10:00    21.190001
2016-06-06 03:00:03+10:00    21.190001
2016-06-06 03:15:03+10:00    21.190001
2016-06-06 03:30:05+10:00    21.190001
2016-06-06 03:45:03+10:00    21.190001
2016-06-06 04:00:04+10:00    21.190001
2016-06-06 04:15:03+10:00    21.190001
2016-06-06 04:30:03+10:00    21.190001
2016-06-06 04:45:02+10:00    21.190001
2016-06-06 05:00:04+10:00    21.190001
2016-06-06 05:15:02+10:00    21.190001
2016-06-06 05:30:04+10:00    21.190001
2016-06-06 05:45:05+10:00    21.190001
2016-06-06 06:00:03+10:00    21.190001
2016-06-06 06:15:03+10:00    21.190001
2016-06-06 06:30:03+10:00    21.190001
2016-06-06 06:45:03+10:00    21.190001
2016-06-06 07:00:03+10:00    21.190001
2016-06-06 07:15:03+10:00    21.190001
                               ...
2016-06-06 16:30:03+10:00    21.190001
2016-06-06 16:45:02+10:00    21.190001
2016-06-06 17:00:03+10:00    21.190001
2016-06-06 17:15:03+10:00    21.190001
2016-06-06 17:30:03+10:00    21.190001
2016-06-06 17:45:03+10:00    21.190001
2016-06-06 18:00:03+10:00    21.190001
2016-06-06 18:15:02+10:00    21.190001
2016-06-06 18:30:03+10:00    21.190001
2016-06-06 18:45:03+10:00    21.190001
2016-06-06 19:00:03+10:00    21.190001
2016-06-06 19:15:03+10:00    21.190001
2016-06-06 19:30:03+10:00    21.190001
2016-06-06 19:45:04+10:00    21.190001
2016-06-06 20:00:03+10:00    21.190001
2016-06-06 20:15:03+10:00    21.190001
2016-06-06 20:30:03+10:00    21.190001
2016-06-06 20:45:03+10:00    21.190001
2016-06-06 21:00:03+10:00    21.190001
2016-06-06 21:15:03+10:00    21.190001
2016-06-06 21:30:04+10:00    21.190001
2016-06-06 21:45:03+10:00    21.190001
2016-06-06 22:00:03+10:00    21.190001
2016-06-06 22:15:03+10:00    21.190001
2016-06-06 22:30:03+10:00    21.190001
2016-06-06 22:45:03+10:00    21.190001
2016-06-06 23:00:05+10:00    21.190001
2016-06-06 23:15:07+10:00    21.190001
2016-06-06 23:30:03+10:00    21.190001
2016-06-06 23:45:03+10:00    21.190001
dtype: float64>
With date slice:
<HisReadSeriesOperation failed>
```

What do you think?

sjlongland, 2016-06-07:
Yep, almost certainly, because `pyhaystack` assumes the double-quotes aren't needed or expected.

ChristianTremblay, 2016-06-07:
Based on the answer of Brian Frank (see http://project-haystack.org/forum/topic/401) we need to modify encoding to include double quotes.

commit 4bcacf4c (2016-06-08): Added verification for empty results and handling for those cases.
Also solves issue15 as we now encode the range into Zinc

This will closes #15, closes #16
closed 2016-06-08 by ChristianTremblay