ChristianTremblay/pyhaystack #29: pyReadLine error

MakhdumS, 2016-09-28:
Hi,

i'm getting an error when i use the grid_format=hszinc.MODE_JSON when setting up the session

Traceback (most recent call last):
  File "C:\Users\makhdum shaikh\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyreadline\console\console.py", line 768, in hook
rapper_23
    res = ensure_str(readline_hook(prompt))
  File "C:\Users\makhdum shaikh\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyreadline\rlmain.py", line 571, in readline
    self._readline_from_keyboard()
  File "C:\Users\makhdum shaikh\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyreadline\rlmain.py", line 536, in _readline_fro
keyboard
    if self._readline_from_keyboard_poll():
  File "C:\Users\makhdum shaikh\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyreadline\rlmain.py", line 556, in _readline_fro
keyboard_poll
    result = self.mode.process_keyevent(event.keyinfo)
  File "C:\Users\makhdum shaikh\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyreadline\modes\emacs.py", line 243, in process_
yeven

ChristianTremblay, 2016-09-29:
We'll need more infos. Can you provide some code ?

sjlongland, 2016-09-29:
I concur, I don't think that's the full stack trae as I don't think anything in pyhaystack cals pyreadline.  This smells like a Python shell going haywire.

Possibly on Windows 10 judging from the mention of "Continuum".

MakhdumS, 2016-10-02:
Managed to resolve this by installing pyreadline

closed 2016-10-02 by MakhdumS