ChristianTremblay/pyhaystack #31: Need support for older Python version (<= 2.6)

ChristianTremblay, 2016-11-01:
We've been asked about the possibility of using older Python version.

ChristianTremblay, 2016-11-01:
There are no plan of providing support for older Python version. We are focusing on Python 3 with actual support for 2.7.

sjlongland, 2016-11-01:
We'll need to audit a lot of code to ensure compatibility, for example, the `logging.Logger` object only added the `getChild` method in Python 2.7:
```
RC=0 stuartl@rikishi ~/vrt/projects/widesky/pyhaystack $ find pyhaystack -type f -name \*.py | xargs grep --line-number getChild
pyhaystack/client/entity/ops/crud.py:30:        self._log = entity._session._log.getChild('update_tags')
pyhaystack/client/ops/his.py:168:        self._log = session._log.getChild('his_read_frame')
pyhaystack/client/ops/his.py:587:        self._log = session._log.getChild('his_write_frame')
pyhaystack/client/ops/grid.py:222:        self._log = session._log.getChild('get_grid.%s' % uri)
pyhaystack/client/ops/grid.py:259:        self._log = session._log.getChild('post_grid.%s' % uri)
pyhaystack/client/session.py:90:            http_args['log'] = log.getChild('http_client')
```

Not sure what the Python 2.6 equivalent would be, but it would not be pretty.

ChristianTremblay, 2016-11-02:
We are open to contributions but we won't fix anytime soon.

closed 2016-11-02 by ChristianTremblay