ChristianTremblay/pyhaystack #32: Improvement: Asynchronous HTTP client

sjlongland, 2016-11-01:
There's one already, it's a "dummy HTTP client" used for unit testing, but I'm thinking a real one would be a nice-to-have at some point in the future.

Possible asynchronous frameworks to target:
* [asyncio](https://docs.python.org/dev/library/asyncio.html) (Python 3.3 and above)
* -[Trollius](http://trollius.readthedocs.io/) (this is a Python 2.x backport of the `tulip` asynchronous framework, the forerunner to `asyncio`, or maybe not as it's now "depreciated")-
* [TornadoWeb](http://www.tornadoweb.org)
* [TwistedMatrix](http://twistedmatrix.com/)
* others?

No rush on this now, the API core is able to support an asynchronous client, this is why it returns a state machine instead of the value when calling methods: the state machine could be running in a separate thread or it could be waiting for control to return to the IO loop.

Supporting a new client is basically a case of writing an implementation in `pyhaystack/client/http` supporting the new HTTP client.

sjlongland, 2018-09-13:
For a HTTP client that behaves like `requests`, this might be worth looking at: https://github.com/littlecodersh/trip

ChristianTremblay, 2019-10-13:
We are not ready yet to put effort in this. I'll close until relevant.

closed 2019-10-13 by ChristianTremblay