ChristianTremblay/pyhaystack #40: pyhaystack authentication fail

thejohnbackes, 2017-04-03:
Hello,

I followed the documentation for SkySpark authentication...

```
 session = SkysparkHaystackSession(uri='https://someDomain.haystack.someProvider.com',
    username='someUserName',
    password='somePassword',
    project='someProject',
    pint = True)

znt = session.find_entity(filter_expr='sensor and power')

print znt.result
```

I receive the following error message:

```
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Users/jbackes/linkedin/infradev/haystack-test/__main__.py", line 14, in <module>
    print znt.result
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/util/state.py", line 97, in result
    self._result.reraise()
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/util/asyncexc.py", line 28, in reraise
    reraise(*self._exc_info)
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/client/ops/entity.py", line 41, in _on_read
    grid = operation.result
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/util/state.py", line 97, in result
    self._result.reraise()
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/util/asyncexc.py", line 28, in reraise
    reraise(*self._exc_info)
  File "/usr/local/lib/python2.7/site-packages/pyhaystack/client/ops/grid.py", line 247, in _do_auth_failed
    raise AuthenticationProblem()
```

ChristianTremblay, 2017-04-04:
If your server uses SCRAM for authentication, you will need what's found under branch feature/scram

Can it be the problem ?

I'm not super familiar with Skyspark but if you login with uri like http://server/ui there are chances that it is the problem. 

Feature scram is in development and we wait to finish the work for Niagara4 login before we release.

ChristianTremblay, 2017-04-26:
@thejohnbackes can you test against the develop branch if it fixes your bugs ?

ChristianTremblay, 2017-05-07:
Our tests are showing us that everything works with 0.92 that will be released tomorrow. If bug re-occurs, we'll reopen an issue

closed 2017-05-07 by ChristianTremblay