ChristianTremblay/pyhaystack #50: AttributeError for SkySpark

p2ya, 2017-05-15:
Hi,
I recently started using Pyhaystack for Skyspark connection by:

import hszinc
from pyhaystack.client.skyspark import SkysparkHaystackSession
session = SkysparkHaystackSession(uri='http://ip',
                                username='username',
                                password='password',
                                project='project',
                                grid_format=hszinc.MODE_JSON
                                )

As suggested on closed issue #14 

I also tried what was on http://pyhaystack.readthedocs.io/en/latest/connect.html#skyspark:
However, I get the following error:

    expect_format=hszinc.MODE_ZINC, multi_grid=False,
AttributeError: 'module' object has no attribute 'MODE_ZINC'

Any idea why this happens and how to resolve it ?

sjlongland, 2017-05-15:
On 15/05/17 11:07, p2ya wrote:
> I also tried what was on
> http://pyhaystack.readthedocs.io/en/latest/connect.html#skyspark:
> However, I get the following error:
> 
> |expect_format=hszinc.MODE_ZINC, multi_grid=False, |
> 
> AttributeError: 'module' object has no attribute 'MODE_ZINC'
> 
> Any idea why this happens and how to resolve it ?

That to me, suggests something is broken with the installation of
`hszinc`.  How did you install everything?
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

p2ya, 2017-05-15:
pip install hszinc

ChristianTremblay, 2017-05-15:
Does it work if you don't specify the format ?

Also, if using version 0.92 (latest), can you test using 

'''
Import pyhaystack
session=pyhaystack.connect("skyspark", uri, user,password)
'''

We may face something related to authentication. The class you are loading isn't meant for Skyspark v3+

Using connect, when specifying "skyspark" the default implementation will use scram authentication.

p2ya, 2017-05-15:
No, it doesn't.

Checked versions:

pip show pyhaystack
---
Name: pyhaystack
Version: 0.92
Location: /usr/local/lib/python2.7/site-packages
Requires: requests, setuptools, pandas, parsimonious, iso8601, hszinc, six, fysom, signalslot, semver

pip show hszinc
---
Name: hszinc
Version: 0.1.1
Location: /usr/local/lib/python2.7/site-packages
Requires: parsimonious, pytz, iso8601, six, pint

Also when I tried the connect() approach I got the following error:

Code:
import pyhaystack
session = pyhaystack.connect(implementation='skyspark',
	                            uri='http://ip',
                                username='username',
                                password='password',
                                project='site'
                                )

Error:

session = pyhaystack.connect(implementation='skyspark',
AttributeError: 'module' object has no attribute 'connect'

p2ya, 2017-05-15:
I am testing that on SkySpark version 2.1.15

ChristianTremblay, 2017-05-15:
Humm... `AttributeError: 'module' object has no attribute 'connect'`is telling me there is a problem with the install..

Can you uninstall pyhaystack and hszonc then reinstall ?
I've faced that kind of problem before but I thought it was because I was alterning between a locl repo and pip install.... anyway.

`pip uninstall pyhsyatck` (multiple time until there is no version left on your system)
`pip uninstall hszinc` (mutliple time again if needed)

then `pip install pyhaytsack` it should install `hszinc`

p2ya, 2017-05-16:
did the uninstall and installed pyhaystack 
but it has the same errors for both direct and connect approach

ronaldsnyder, 2017-05-16:
Did you install using sudo?  May need to install using virtualenv

p2ya, 2017-05-16:
tried the sudo and same error

p2ya, 2017-05-16:
is there any chance that you test it for SkySpark and see if it work ?

sjlongland, 2017-05-17:
I personally don't have access to SkySpark… but my hunch is there's a dependent Python module (of `hszinc`) that isn't loading for some reason.

That is why there's an `AttributeError`.

ronaldsnyder, 2017-05-17:
Are you using a Mac?  This sounds similar to issue 34, https://github.com/ChristianTremblay/pyhaystack/issues/34

These are the steps that @ChristianTremblay gave me before that fixed it.

It is probably related to some kind of permission of the "system's python" version.
To be able to make it work, you will need to use a virtual env that will isolate you from the system's python.
Here is a simple guide : http://docs.python-guide.org/en/latest/dev/virtualenvs/
You will need to :

sudo pip install virtualenv
mkdir your project folder
cd project
virtualenv venv
source venv/bin/activate

Once you are in your virtual env DO NOT use sudo to pip install.
(in fact, this is the part that made me think of permission issue as I read somewhere that we should never sudo pip install anything)

So now you are in your virtual env (it's in parenthesis in the console) and you

pip install requests
pip install hszinc
pip install pyhaystack
(note that this time you won't see any weird message when trying to install pandas and you need xcode to perform the install....)
You are now able to :

import hszinc
hszinc.MODE_ZINC

from pyhaystack.client.skyspark import SkysparkHaystackSession
# won't raise an error

ChristianTremblay, 2017-05-17:
I've also had some problem with pip unable to uninstall things... you could also check to update pip : 
`python -m pip install --upgrade pip`
Then uninstall pyhaystack 
the `attribute error`is not normal as it tells that function connect doesn't exist... 

I've just tested a Skyspark v2 connection : 
```
import pyhaystack

session = pyhaystack.connect('skyspark2', uri='http://ip',
                                username='user',
                                password='pw',
                                project = 'myProject')

session.about().result
```
Please note that implementation for old Skyspark version is `skyspark2` and we chose 'skyspark`as the shortcut for v3+ which will probably become the "by default" option anyway.

This also led to a bug discovered with bracket search... guess I'll open an issue and publish a hot fix....

p2ya, 2017-05-17:
Hi Ron
Worked like a charm, thanks for the help.
I will carry on using the package.

ChristianTremblay, 2017-05-17:
I will add this to our documentation. It will probably help others.

ChristianTremblay, 2017-05-17:
Documentation has been updated in develop branch

closed 2017-05-17 by ChristianTremblay