ChristianTremblay/pyhaystack #53: Call SkySpark Functions

p2ya, 2017-06-27:
I wanted to ask if we can call developed functions in SkySpark with specific inputs through pyhaystack and get the result back ?

ChristianTremblay, 2017-06-30:
I will need more details. I know very little from skyspark and I don't have access to a site where I can make tests.
Is there a way to call functions from REST API ?

sjlongland, 2017-07-01:
I think I helped @pierresigwalt with this a few weeks back.  It's an `exec` call correct?

We managed to do it with a call like this:
```
self.session._get_grid('eval',expr='expr=navFilter("sparkTarget",readAll(equip)).sparks(2017-02).sparkHis', callback=lambda *a, **k: None)
```
Not friendly, we're calling private methods here… but doable.

We will have to add an `evalexpr()` endpoint to the SkySpark classes methinks… that'll wrap the above code up in a nice easy-to-call function.

pierresigwalt, 2017-07-01:
Exactly,

I was able to request particular data using the library.

Code sample:

def get_eval(self, arg_expr):
        url = '/eval?expr=%s' % arg_expr
        result = self.session._get_grid(url, callback=lambda *a, **k: None)
        return result

ChristianTremblay, 2017-07-05:
@pierresigwalt @sjlongland 
I think this should be implemented in skyspark client, and documented.
Skyspark is widely used and custom functions compliance inside pyhaystack can be a game changer for someone considering using python and Skyspark vs only Skyspark. Pierre's function is clear.

sjlongland, 2017-07-05:
On 05/07/17 11:14, Christian Tremblay wrote:
> @pierresigwalt <https://github.com/pierresigwalt> @sjlongland
> <https://github.com/sjlongland>
> I think this should be implemented in skyspark client, and documented.
> Skyspark is widely used and [...]

Agreed. :-)

-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
   ...it's backed up on a tape somewhere.

ChristianTremblay, 2017-07-05:
As it is a very small modification, I didn't create a feature branch. 
Please have a look to https://github.com/ChristianTremblay/pyhaystack/commit/de81823a0c9e82621fd4ae86101842ec810b34a0

ChristianTremblay, 2017-07-06:
I corrected many little errors I made yesterday in the hurry....
I've been able to test 
```
session.get_eval("readAll(site)")
```

It works great now : 
https://github.com/ChristianTremblay/pyhaystack/commit/469bde306dd1751837b485374f37c09e9fe93b45

ChristianTremblay, 2017-07-06:
@p2ya can you confirm that we are responding to your need with `get_eval` function. If so, I will close the issue.

Thanks for your interest in pyhaystack

p2ya, 2017-07-26:
Thanks for the update, however I get the following error:
AttributeError: 'SkysparkHaystackSession' object has no attribute 'get_eval'

pierresigwalt, 2017-07-26:
Hum,

What I checked is still in the develop branch (get_eval feature) so if you want to test you may use that branch instead of the master.

Christian will be able to confirm but I think it has not been pushed yet to Master.

Thank you

p2ya, 2017-07-26:
Hi @ChristianTremblay , 
can you please let me know how I can test this or if it is possible to push it to the master branch

sjlongland, 2017-07-26:
Try:

```
pip install git+https://github.com/ChristianTremblay/pyhaystack.git@develop
```

p2ya, 2017-07-27:
Installed and tested, I get the following error:
<GetGridOperation failed>
When I try:
op = session.get_eval("readAll(site)") 
op.wait()
print op

p2ya, 2017-07-28:
Hi @sjlongland,
Any idea why I get this error and how it can be handled ?

sjlongland, 2017-07-28:
Not the foggiest.  I might suggest you use `print op.result` not `print op`.

The former will print the actual result, or raise the error caused, the latter just returns a string representation of the state machine which simply tells us that it didn't work, it doesn't tell us why.

p2ya, 2017-07-28:
please find it as:
pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 404 Client Error: Not Found for url: http://IP/api/sites**//**eval?expr=readAll(site)] 404

it seems that the issue is double //

sjlongland, 2017-07-28:
That appears to suggest the server doesn't support `eval`.

p2ya, 2017-07-28:
I can get the result when I run the same request with one "/" though as:
http://IP/api/sites/eval?expr=readAll(site)
rather than
http://IP/api/sites//eval?expr=readAll(site)

p2ya, 2017-07-31:
Hi @ChristianTremblay ,
you reckon this can be fixed ?

ChristianTremblay, 2017-07-31:
This won't be fixed right now as I'm in the middle of my summer vacations. But I will give a high priority to this as soon as I come back.

p2ya, 2017-07-31:
Hi @ChristianTremblay, no worries. Hope you have a nice holiday and looking forward to hear back from you to test this again and bring it into action.

p2ya, 2017-08-15:
the latest update has fixed the issue and I can call eval function.

closed 2017-08-15 by p2ya