ChristianTremblay/pyhaystack #54: <FindEntityOperation failed>

p2ya, 2017-06-28:
I get the error "FindEntityOperation failed" when I run:
op = session.find_entity(filter_expr="zone and temp and sensor")
however it return correct values when I run:
op = session.find_entity(filter_expr="ahu and equip")

This is through SkysparkHaystackSession

sjlongland, 2017-06-29:
On 28/06/17 10:26, p2ya wrote:
> I get the error "FindEntityOperation failed" when I run:
> op = session.find_entity(filter_expr="zone and temp and sensor")
> however it return correct values when I run:
> op = session.find_entity(filter_expr="ahu and equip")
> 
> This is through SkysparkHaystackSession

If you inspect op.result, you should see the exception that triggered
the failure.

-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

p2ya, 2017-06-30:
Hi Stuart,
I test that for two points by reading their SkySpark IDs, op = session.find_entity(filter_expr= """id==@...""")
one works nicely but one throws an error as follow through op.result:

parsimonious.exceptions.IncompleteParseError: Rule 'grid' matched in its entirety, but it didn't consume all the text. The non-matching portion of the text begins with '@20a7e02c-ff2c81e7 "' (line 3, column 1).

sjlongland, 2017-06-30:
On 30/06/17 10:57, p2ya wrote:
> Hi Stuart,
> I test that for two points by reading their SkySpark IDs, op =
> session.find_entity(filter_expr= """id==@...""")
> one works nicely but one throws an error as follow through op.result: [...]

In other words, the server returned more than what the client was 
bargaining for.

I wonder if you could try creating the session with a logger and try the 
action again?

Something like the following should do the trick:

import logging
import pyhaystack

logger.basicConfig(level=logging.DEBUG, filename='/tmp/skyspark.log')

session = pyhaystack.connect(your, args, here, \
     log=logging.getLogger('pyhaystack'), http_args=dict(debug=true))

That'll create `skyspark.log` in your `/tmp` directory.  In the log 
should be a raw dump of exactly what Skyspark sent in its reply and we 
can amend `hszinc`'s parsing rules to accommodate it.
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
   ...it's backed up on a tape somewhere.

p2ya, 2017-06-30:
Do you have a working example which uses the log file ?
I get couple of errors implementing it

candronikos, 2017-06-30:
Just to follow up after @p2ya . I implemented the code you suggested and the result can be seen below.

```
DEBUG:requests.packages.urllib3.connectionpool:Starting new HTTP connection (1): host.com
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "GET /auth/ss_project/api?user HTTP/1.1" 200 141
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "POST /auth/ss_project/api?user HTTP/1.1" 200 66
DEBUG:pyhaystack.client.SkysparkHaystackSession.get_grid.read:Authenticated, trying again
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "GET /api/ss_project/read?filter=id+%3D%3D+1c264c3d-ccfc92a4 HTTP/1.1" 200 None
DEBUG:pyhaystack.client.SkysparkHaystackSession.find_entity:Received grid: <Grid>
	Metadata: MetadataObject{'projName'='ss_project'}
	Columns:
		id
		analytics
		cmd
		disMacro
		equipRef
		his
		hisEnd
		hisEndVal
		hisId
		hisOnWrite
		hisSize
		hisStart
		hisStatus
		hot
		kind
		levelRef
		navName
		obixConnRef
		obixHis
		point
		regionRef
		siteRef
		stateRef
		tz
		unit
		valve
		vav
		water
		mod
	Row    0: id=Ref('1c264c3d-ccfc92a4', 'CLIENT - Site VAV2 8C HWV', True), analytics=MARKER, cmd=MARKER, disMacro='$equipRef $navName', equipRef=Ref('1c1258a2-6dd7daf9', 'CLIENT - Site VAV2 8C', True), his=MARKER, hisEnd=datetime.datetime(2017, 6, 30, 13, 15, 9, tzinfo=<DstTzInfo 'Australia/Hobart' AEST+10:00:00 STD>), hisEndVal=BasicQuantity(0.0, '%'), hisId='0.530', hisOnWrite='fixObixTz', hisSize=104243.0, hisStart=datetime.datetime(2014, 6, 22, 11, 15, 10, tzinfo=<DstTzInfo 'Australia/Hobart' AEST+10:00:00 STD>), hisStatus='ok', hot=MARKER, kind='Number', levelRef=Ref('1eaac91f-8b1434fa', 'CLIENT - Site Level 2', True), navName='HWV', obixConnRef=Ref('1be5ab92-4871f8e1', 'CLIENT - Site', True), obixHis=Uri('/obix/histories/St_John_St_79_G_L3/VAV2_8C_HWV/'), point=MARKER, regionRef=Ref('1ea9d962-bb8f33b3', 'North Tasmania', True), siteRef=Ref('1be9855e-7ab53b9f', 'CLIENT - Site', True), stateRef=Ref('1ea9d50b-0054078a', 'TAS', True), tz='Hobart', unit='%', valve=MARKER, vav=MARKER, water=MARKER, mod=datetime.datetime(2017, 4, 21, 3, 57, 17, 644000, tzinfo=<iso8601.Utc>)
</Grid>
DEBUG:requests.packages.urllib3.connectionpool:Starting new HTTP connection (1): host.com
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "GET /auth/ss_project/api?user HTTP/1.1" 200 141
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "POST /auth/ss_project/api?user HTTP/1.1" 200 66
DEBUG:pyhaystack.client.SkysparkHaystackSession.get_grid.read:Authenticated, trying again
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "GET /api/ss_project/read?filter=id+%3D%3D+20a7e02c-ff2c81e7 HTTP/1.1" 200 None
DEBUG:pyhaystack.client.SkysparkHaystackSession.get_grid.read:Parse fails
Traceback (most recent call last):
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/pyhaystack/pyhaystack/client/ops/grid.py", line 217, in _on_response
    decoded = hszinc.parse(body, mode=hszinc.MODE_ZINC)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 63, in parse
    return list(map(_parse, GRID_SEP.split(grid_str.rstrip())))
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 70, in parse_grid
    parsed = zinc_grammar.parse(grid_str)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/grammar.py", line 123, in parse
    return self.default_rule.parse(text, pos=pos)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/expressions.py", line 112, in parse
    raise IncompleteParseError(text, node.end, self)
parsimonious.exceptions.IncompleteParseError: Rule 'grid' matched in its entirety, but it didn't consume all the text. The non-matching portion of the text begins with '@20a7e02c-ff2c81e7 "' (line 3, column 1).
DEBUG:requests.packages.urllib3.connectionpool:http://host.com:80 "GET /api/ss_project/read?filter=id+%3D%3D+20a7e02c-ff2c81e7 HTTP/1.1" 200 None
DEBUG:pyhaystack.client.SkysparkHaystackSession.get_grid.read:Parse fails
Traceback (most recent call last):
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/pyhaystack/pyhaystack/client/ops/grid.py", line 217, in _on_response
    decoded = hszinc.parse(body, mode=hszinc.MODE_ZINC)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 63, in parse
    return list(map(_parse, GRID_SEP.split(grid_str.rstrip())))
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 70, in parse_grid
    parsed = zinc_grammar.parse(grid_str)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/grammar.py", line 123, in parse
[... 155 lines cut ...]
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 63, in parse
    return list(map(_parse, GRID_SEP.split(grid_str.rstrip())))
  File "/home/user/Documents/projects/pyhaystack_test/.venv/src/hszinc/hszinc/parser.py", line 70, in parse_grid
    parsed = zinc_grammar.parse(grid_str)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/grammar.py", line 123, in parse
    return self.default_rule.parse(text, pos=pos)
[... cut ...]
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/grammar.py", line 123, in parse
    return self.default_rule.parse(text, pos=pos)
  File "/home/user/Documents/projects/pyhaystack_test/.venv/lib/python3.6/site-packages/parsimonious/expressions.py", line 112, in parse
    raise IncompleteParseError(text, node.end, self)
parsimonious.exceptions.IncompleteParseError: Rule 'grid' matched in its entirety, but it didn't consume all the text. The non-matching portion of the text begins with '@20a7e02c-ff2c81e7 "' (line 3, column 1).
```

sjlongland, 2017-06-30:
Hmm, I thought that should have emitted the response back in handling the parse error.  Okay, there's a debug statement that should be added for future tracing purposes.

Just thinking though, SkySpark supports JSON grid formats, maybe try putting `grid_format=hszinc.MODE_JSON` in your session arguments… that will switch it to use JSON instead of ZINC mode.  ZINC is used with nHaystack as it doesn't implement JSON properly, everything else is new enough to have JSON grids standardised.

ChristianTremblay, 2017-06-30:
I agree... and i could result in setting default mode to JSON for skyspark eventually...

p2ya, 2017-07-01:
changing grid format worked

closed 2017-08-08 by ChristianTremblay