ChristianTremblay/pyhaystack #61: pyhaystack.connect failed with Niagara 14.4

ChristianTremblay, 2018-03-29:
Results in Authentication failure

Using : 
```
from pyhaystack.client.niagara import Niagara4HaystackSession
import requests
from requests.packages.urllib3.exceptions import InsecureRequestWarning
requests.packages.urllib3.disable_warnings(InsecureRequestWarning)
session = Niagara4HaystackSession(uri='https://192.168.1.2', 
                                  username='user', 
                                  password='password',
                                  http_args={'debug':True, 'tls_verify':False})
            
session.about().result
```
works

sjlongland, 2018-03-29:
I'll bite, where did you get Niagara 14.4 from?

ChristianTremblay, 2018-03-29:
Oups ! JCI Workbench 14.4 which is Niagara 4.4.73.24

ChristianTremblay, 2018-04-05:
Fixed in 0.95.6 (develop)

closed 2018-04-05 by ChristianTremblay