ChristianTremblay/pyhaystack #64: ValueError in grid is symptomatic of a disconnection from server

ChristianTremblay, 2018-05-08:
Getting this error should trig a reconnection attempt when content_type is text/html

ChristianTremblay, 2018-05-08:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\client\ops\grid.py in _on_response(self, response)
    286                 # We don't recognise this!
--> 287                 raise ValueError('Unrecognised content type %s' % content_type)
    288 

ValueError: Unrecognised content type text/html

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\client\ops\grid.py in _on_response(self, response)
    286                 # We don't recognise this!
--> 287                 raise ValueError('Unrecognised content type %s' % content_type)
    288 

ValueError: Unrecognised content type text/html

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-6-148496b3bee8> in <module>()
      1 for floor in range(3,17):
      2     if floor != 13:
----> 3         check_floor(floor, '2018-05-04')

<ipython-input-5-433aada04b66> in check_floor(floor, date)
      2     try:
      3         request = 'air and zone and temp and sensor and equipRef->floorName=="Floor {}"'.format(floor)
----> 4         room_zone_temp = session.find_entity(filter_expr=request).result
      5         room_zone_temp_df = session.his_read_frame(room_zone_temp, rng=date).result
      6         for each in room_zone_temp_df.items():

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\util\state.py in result(self)
     95 
     96         if self.is_failed:
---> 97             self._result.reraise()
     98 
     99         if not self._result_copy:

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\util\asyncexc.py in reraise(self)
     26 
     27     def reraise(self):
---> 28         reraise(*self._exc_info)

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\six.py in reraise(tp, value, tb)
    691             if value.__traceback__ is not tb:
    692                 raise value.with_traceback(tb)
--> 693             raise value
    694         finally:
    695             value = None

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\client\ops\entity.py in _on_read(self, operation, **kwargs)
     39         try:
     40             # See if the read succeeded.
---> 41             grid = operation.result
     42             self._log.debug('Received grid: %s', grid)
     43 

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\util\state.py in result(self)
     95 
     96         if self.is_failed:
---> 97             self._result.reraise()
     98 
     99         if not self._result_copy:

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\util\asyncexc.py in reraise(self)
     26 
     27     def reraise(self):
---> 28         reraise(*self._exc_info)

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\six.py in reraise(tp, value, tb)
    691             if value.__traceback__ is not tb:
    692                 raise value.with_traceback(tb)
--> 693             raise value
    694         finally:
    695             value = None

C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3\lib\site-packages\pyhaystack-0.92.6-py3.6.egg\pyhaystack\client\ops\grid.py in _on_response(self, response)
    285             else:
    286                 # We don't recognise this!
--> 287                 raise ValueError('Unrecognised content type %s' % content_type)
    288 
    289             # Check for exceptions

ValueError: Unrecognised content type text/html

```

ChristianTremblay, 2018-05-08:
@sjlongland 
Would this be a valid way to trig a retry

`if content_type in ('text/zinc', 'text/plain'):
                # We have been given a grid in ZINC format.
                decoded = hszinc.parse(body, mode=hszinc.MODE_ZINC)
            elif content_type == 'application/json':
                # We have been given a grid in JSON format.
                decoded = [hszinc.parse(body, mode=hszinc.MODE_JSON)]
            elif content_type in ('text/html'):
                # We probably fell back to a login screen after auto logoff.
                self._state_machine.exception(AsynchronousException())
            else:
                # We don't recognise this!
                raise ValueError('Unrecognised content type %s' % content_type)`

sjlongland, 2018-05-08:
Well, it'd have to be `('text/html',)`, because otherwise the brackets do nothing.  We need to have a method on the session object that cleans up the log-in session when this happens.

ChristianTremblay, 2020-05-28:
Using json with Niagara fixed this.

closed 2020-05-28 by ChristianTremblay