ChristianTremblay/pyhaystack #66: Question: SkySpark 3 Authentication

pouyaghadimi, 2018-07-20:
Does pyhaystack authentication work on SkySpark 3 ?

sjlongland, 2018-07-20:
It should do.  I can't test myself, but my understanding is that it should be working with both Skyspark 2 and 3.

ChristianTremblay, 2018-07-20:
Skyspark 3 is the default for pyhaystack
To log into Skyspark 2 you must specify it

```
IMPLEMENTATION_ALIAS = {
        'niagara-ax':   'niagara.NiagaraHaystackSession',
        'ax':           'niagara.NiagaraHaystackSession',
        'niagara4':     'niagara.Niagara4HaystackSession',
        'n4':           'niagara.Niagara4HaystackSession',
        'skyspark2':    'skyspark.SkysparkHaystackSession',
        'skyspark':     'skyspark.SkysparkScramHaystackSession',
        'widesky':      'widesky.WideskyHaystackSession',
}
```

closed 2018-07-20 by ChristianTremblay