ChristianTremblay/pyhaystack #67: Authentication problem with Niagara 4.6

tybeady, 2018-08-24:
I have been getting the following error when trying to connect to a JACE 8000:

  File "C:\ProgramData\Anaconda3\lib\site-packages\pyhaystack\client\ops\grid.py", line 133, in _do_auth_failed
    raise AuthenticationProblem()

AuthenticationProblem

I have set up the user as a basicHTTP user but I still can't connect. After watching the traffic it looks like the JACE sends back a Hello handshake but then pyhaystack just gives up and tries again.

ChristianTremblay, 2018-08-24:
I don't have access yet to 4.6..... but here is what I do with 4.4
The user I use is configured with digest using all rights I need (html5 web Profile) and for get_bql to work, we need the hx_view checked in the web profile (going from memory here... may be slightly different name.

Then I use this script to log in : 
```
from pyhaystack.client.niagara import Niagara4HaystackSession
session = Niagara4HaystackSession(uri='https://192.168.1.149',
                                 username=user,
                                 password=password,
                                 http_args={'debug':True, 'tls_verify':False, 'insecure_requests_warning':False})
```

Hope it'll help

ChristianTremblay, 2019-02-13:
@tybeady It work on my side with 4.6 using SSL. Is this still a problem for you ?

sjlongland, 2019-02-13:
Really, we should pull our fingers out and come up with a more helpful error message… when I wrote that code I didn't bother trying to add code to diagnose it further and it does make things difficult at times.

The `http_args={'debug': True}`, coupled with setting up `logging` (do `import logging; logging.basicConfig(level=logging.DEBUG)` prior to attempting to connect) can help diagnose a lot of issues.

closed 2019-03-30 by ChristianTremblay