ChristianTremblay/pyhaystack #70: (Question?): Closing PyHaystack sessions on both client and server.

adam-rysanek, 2019-04-10:
We are encountering with how to execute PyHaystack frequently on a client machine in a way that does not burden our SkySpark server. Our client machine executes a python script every 10 minutes. This script is being used to pull data from our Skyspark server and push data to another data server we are working with. The framework of our script is as follows:

'''
session = pyhaystack.connect('skyspark',
                            uri='https://serveraddress',
                            username='user',
                            password='password',
                            project = 'project',
                            pint=True,
                            grid_format=hszinc.MODE_JSON,
                            http_args={'debug':False},
                             )

data_pull = session.get_eval("apiDataTransfer()")
data_pull.wait()
data_json=remove_null_bool(json.loads(hszinc.dump(data_pull.result, mode=hszinc.MODE_JSON)))

do other things until end of file

'''

This script runs every 10 minutes on the client machine (as a recurring cron job). The script takes about 30 seconds to run, in terms of connecting to the Skyspark server and pulling data from the "APIdatatransfer()" call.

We have come to learn that by running the script this way, on the SkySpark server opens its own session each time the client connects to it. Each session is left open for a default period of 3 hours as the Python script does not provide any indication to the server that the session should close. As a result, our run-every-10-minutes script leads to opening 180 sessions on teh server and is unnecessarily overloading the server.

Whilst we can fix this problem by running our every-10-minutes process as a recurring loop within a single Python file (and thus only 1 session), I would like to know if there is a way to disconnect a session properly through the PyHaystack library.

In the same way that 'pyhaystack.connect' establishes a connection, is there anything to the effect of 'pyhaystack.disconnect'?

Many thanks for your feedback.

sjlongland, 2019-04-10:
On 11/4/19 7:28 am, adam-rysanek wrote:
> In the same way that 'pyhaystack.connect' establishes a connection, is
> there anything to the effect of 'pyhaystack.disconnect'?

There isn't… and I have no idea how that is done in SkySpark.  It's been
something in the back of my mind that I should implement something like
it for WideSky (it's simply a DELETE REST call to /oauth2/token IIRC)
but then again, WideSky's behaviour is different in this regard.

nHaystack on Niagara AX just uses basic authentication, and so doesn't
strictly have a "session" as such -- and that's what PyHaystack was
modelled on originally.  Niagara4 (with SCRAM) came later and again, I
haven't researched how the session is closed off.

adam-rysanek, 2019-04-11:
Thanks very much @sjlongland . For now we will build our python script with a recurring loop within a single session. I think programmatically closing a session does have value, however, but sorry that I am also not well versed enough on the SkySpark side to offer help in developing this.

ChristianTremblay, 2019-04-13:
Looking at Skyspark doc, there is no direct instruction to logout from the API... But I'm wondering if we could just point to something like : 
`https://server/logout` when we need to explicitly logout... same would apply with Niagara

ChristianTremblay, 2019-05-21:
@adam-rysanek : any update on that ?

ChristianTremblay, 2019-10-13:
Closing, if you need something else, please reopen

closed 2019-10-13 by ChristianTremblay

sjlongland, 2022-02-18:
https://project-haystack.org/forum/topic/968 Looks like there's a proposal to this effect.