ChristianTremblay/pyhaystack #73: support for Kodaro haystack driver

bergey, 2019-05-23:
Does pyhaystack support the [Kodaro haystack driver](https://www.kodaro.com/drivers)?  I don't see it mentioned in [the docs](https://pyhaystack.readthedocs.io/en/latest/connect.html), but would like to confirm.

Assuming this is a feature request, can the maintainers give me a sense of how hard this would be, what level of interest they have in the feature, and what support they can give if I decide to have a go?

I'm running into some Niagara devices running this plugin.  It's not yet clear whether installing the N4 plugin is an option at these sites.

ChristianTremblay, 2019-05-24:
I never had the chance to try. But from what I understand... I think we could easily start from the niagara4 client which handle the authentication part... then, we need to know what uri Kodaro uses for their driver to make requests.
Are you in a position to help  ? Provide a connection to a Kodaro instance ?

bergey, 2019-05-24:
At the current project I was able to get NHaystack installed instead.  If it comes up again I will see if I can make the device available for extending pyhaystack.  Thanks!

ChristianTremblay, 2019-05-27:
Please reopen if needed in the future.

closed 2019-05-27 by ChristianTremblay