ChristianTremblay/pyhaystack #74: Skyspark with pyhaystack to create and modify points / tags

Ckoshnick, 2019-05-29:
Hello,

I'm a new user of skyspark and we're interested in integrating our campus' sites and historical data through the REST API or by using the pyhaystack library if that supports what we're trying to accomplish.

So far I've been successful in connecting to my server with the pyhaystack client and been able to read tags with operations like op = session.find_entity(filter_expr='temp')

an example sensor looks like this:

    t1.tags
    Out[20]: {air, disMacro='$equipRef $navName', equipRef=Ref('p:ucdavis:r:246332ce-cd5328e0', 'GBSF VAV6512', True), his, kind='Number', locationRef=Ref('p:ucdavis:r:24638759-cc1a3b2a', 'Floor 5', True), mod=datetime.datetime(2019, 5, 7, 17, 4, 17, 282000, tzinfo=datetime.timezone.utc), navName='Zone Air Temp', point, sensor, siteRef=Ref('p:ucdavis:r:24632687-0e4bd94c', 'GBSF', True), temp, tz='Los_Angeles', unit='°F'}

All is fine and well, except when I want to modify the tag list:

    t1.tags['faketag1'] = None

    Traceback (most recent call last):
    File "<ipython-input-27-4eb0de79124a>", line 1, in <module>
    t1.tags['faketag1'] = None

    TypeError: 'ReadOnlyEntityTags' object does not support item assignment

Is there a setting in my skyspark server or the pyhaystack client that is ensuring that all of my tags are of type "ReadOnlyEntityTags" and not "MutableEntityTags"

This is the section of the pyhaystack documentation I was following along with. https://pyhaystack.readthedocs.io/en/latest/tags.html#adding-changing-and-deleting-tags

Thanks!

sjlongland, 2019-05-30:
To the best of my knowledge, the only server that implements CRUD ops (and thus can use `MutableEntityTags`) is WideSky.

If you know of a way to perform CRUD ops for SkySpark and are willing to provide a patch, we can consider reviewing it, but right now I do not know of a way to modify the asset model in a SkySpark instance which is why the SkySpark code returns a `ReadOnlyEntityTags` instance.

ChristianTremblay, 2019-10-13:
CLosing. If you still need this, please reopen

closed 2019-10-13 by ChristianTremblay