ChristianTremblay/pyhaystack #75: Gitter/Jar Files

bbartling, 2019-06-24:
Hi Christian,

Do you use gitter still for pyhaystack help? I cant find it thru googling...

Could you give me a tip on where I can find the Jar files online to implement nHaystack on an N4 server? Also would you know if the server running 4.2, can I use a 4.7 workbench to put nHaystack jar files on the server thru the software manager on a platform connection?

Thanks,
Ben

sjlongland, 2019-06-24:
On 25/6/19 12:58 am, bbartling wrote:
> Hi Christian,
>
> Do you use gitter still for pyhaystack help? I cant find it thru googling...
>
> Could you give me a tip on where I can find the Jar files online to
> implement nHaystack on an N4 server? [...]

It's on Bitbucket (which is an utter turd for discovering repositories…
go to the front page and it's all about *them* and not the projects they
host).

https://bitbucket.org/richiemac_77/nhaystack/downloads/

I think with Workbench too… it has a happy knack of wanting to update
the version of Niagara on the Jace you're connected to to match the
version of Workbench.  That said, I'm no Niagara expert -- know enough
to be dangerous, and how to hack drivers, but that's about it.
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

ChristianTremblay, 2019-06-25:
You should be ok with your WB to install the file. Right click, install then commit.

bbartling, 2019-06-25:
Have you ever ran into this error during the 'commit' in platform connection to the station? Its running 4.2 and I using 4.7 workbench, could that be the issue? Java for ya :(

```
com.tridium.platform.daemon.DaemonResponseException: Request to platform daemon at 10.30.1.143 failed:

   at com.tridium.platform.daemon.DaemonFileUtil.transfer(DaemonFileUtil.java:650)
   at com.tridium.install.InstallScenario.commit(InstallScenario.java:1270)
   at com.tridium.install.InstallScenario.commit(InstallScenario.java:1052)
   at com.tridium.platDaemon.ui.softwaremanager.CommitCommand.doInvokeAsync(CommitCommand.java:182)
   at com.tridium.platform.ui.util.AsyncCommand$1.run(AsyncCommand.java:83)
```

[image]

bbartling, 2019-06-26:
Just for the heck of it I tried, workbench 4.2, and there appears to be a laundry list of unsatisfied dependacies. Ill talk the sales guy to see if the customer can upgrade there site ... Unless and older version of nHaystck would work?

[image]

sjlongland, 2019-06-27:
I think nHaystack had big issues running on Niagara 4.3 and earlier one of which being that the web framework completely overrode the authentication mechanisms in any serverlets you tried to run on the web server (including nHaystack).

@ChristianTremblay had some real fun un-picking *that* mess!  I think `pyhaystack` now works with such a set-up… but given nHaystack no longer seems to support earlier Niagara releases, that might be a moot point.

(Also worth pointing out, this *technically* isn't a support channel for nHaystack… that officially is [over there](https://bitbucket.org/richiemac_77/nhaystack/issues?status=new&status=open). :-))

bbartling, 2019-06-27:
Thanks guys for the help...

bbartling, 2019-06-27:
Is gitter still used for tips and tricks?

closed 2019-06-27 by bbartling

sjlongland, 2019-06-27:
It is… I haven't been monitoring it much as I've been flat out with work and other projects.

bbartling, 2019-06-27:
I can't find the gitter account for pyhaystack online... Can someone send me a link? I was just hoping to dig up some old notes posted on there :^) thanks

sjlongland, 2019-06-27:
Well, on the [project README page](https://github.com/ChristianTremblay/pyhaystack/) there's a link to the [Gitter chat-room](https://gitter.im/ChristianTremblay/pyhaystack?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=body_badge).

[image]