ChristianTremblay/pyhaystack #76: HaystackError: java.lang.ClassCastExceptio

bbartling, 2019-07-03:
Would there be anyway someone could give me a tip on if I am doing anything wrong? To make a long story short a customer is interested in reports above & beyond the Niagara reporting service capabilities but they are at an N4 revision too low to give them an example of pyhaystack/pandas/matplotlib... (4.2, non TLS)

So I am hoping to test this out/give them an example from a different job/server which is at revision 4.7 but also secure TLS.

Im experimenting in Jupyter lab notebooks on Python 3.7. I can print the NAV file of the N4 server:
[image]

But attempting this code Ill get a long error:
```
# Retrieve some points
op = session.find_entity(filter_expr='hwst')
op.wait()
hwst = op.result
# Read today's history for all found points
op = session.his_read_frame(hwst, rng= 'today')
op.wait()
b = op.result
b
```
[image]

Full trace back:
```
---------------------------------------------------------------------------
HaystackError                             Traceback (most recent call last)
<ipython-input-5-f1f36714fb3f> in <module>
----> 1 hwst = op.result

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\util\state.py in result(self)
     95 
     96         if self.is_failed:
---> 97             self._result.reraise()
     98 
     99         if not self._result_copy:

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\util\asyncexc.py in reraise(self)
     26 
     27     def reraise(self):
---> 28         reraise(*self._exc_info)

~\AppData\Local\Continuum\anaconda3\lib\site-packages\six.py in reraise(tp, value, tb)
    691             if value.__traceback__ is not tb:
    692                 raise value.with_traceback(tb)
--> 693             raise value
    694         finally:
    695             value = None

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\client\ops\entity.py in _on_read(self, operation, **kwargs)
     39         try:
     40             # See if the read succeeded.
---> 41             grid = operation.result
     42             self._log.debug('Received grid: %s', grid)
     43 

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\util\state.py in result(self)
     95 
     96         if self.is_failed:
---> 97             self._result.reraise()
     98 
     99         if not self._result_copy:

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\util\asyncexc.py in reraise(self)
     26 
     27     def reraise(self):
---> 28         reraise(*self._exc_info)

~\AppData\Local\Continuum\anaconda3\lib\site-packages\six.py in reraise(tp, value, tb)
    691             if value.__traceback__ is not tb:
    692                 raise value.with_traceback(tb)
--> 693             raise value
    694         finally:
    695             value = None

~\AppData\Local\Continuum\anaconda3\lib\site-packages\pyhaystack\client\ops\grid.py in _check_err(grid)
    296                         raise HaystackError(
    297                                 grid.metadata.get('dis', 'Unknown Error'),
--> 298                                 grid.metadata.get('traceback', None))
    299                     return grid
    300                 except:

HaystackError: java.lang.ClassCastException
```

Any tips greatly appreciated, as I mentioned this before its just to get sample of pyhaystack capabilities to use on a different site that is non TLS. (Get the customer to upgrade to 4.7). There is another group in house that does Skysparks that had this nHaystack service running already and I just added the `hwst` tag as a test trial.

[image]
[image]
[image]

ChristianTremblay, 2019-07-03:
Have you tried "Rebuild Cache" on the nhaystack service ?

bbartling, 2019-07-03:
In workbench, I did the right click command on the nhaystack service where it rebuilt the cache successfully... Still same issues tho when re-running the Jupyter lab code tho..

There is a different group in a different office (same company) that does Sky sparks... And I think they route points to a cloud based n4 server then to Sky Foundry somehow... Just FYI, it would be nice to try and connect to the Sky sparks server but I am not very involved with that group very much

[image]

bbartling, 2019-07-03:
The nHaystack service was already running on this N4 server before me... Which I found sort of interesting

ChristianTremblay, 2019-07-03:
Using the Query view of the nhaystack service module, can you find hswt ?

bbartling, 2019-07-03:
Can you give me a tip on how to do that?

ChristianTremblay, 2019-07-03:
Using the nhaystack palette...
Under config, I usually put the `site` object (see palette)...and configure it.
Under the controller (typically) I add a `equip` object (again see the palette)...and configure it. Be sure to reference the equipment to the site you created.

Once done, `rebuild cache`
Every time you modify something, `rebuild cache`

The `Query View` is just another view on the service. You can write a filter expression hit enter and you'll get results.

bbartling, 2019-07-09:
Would anyone be able to give me any tips if I am missing something? I did add the site and an equip tag. The HwsT point is where my point tag is implemented:
[image]
[image]

I can print the about & nav file just fine.. Also trying setting `pint=True`
[image]
[image]

But trying the `site = session.site` this also brings up the same exact Java error with full traceback I pasted in previously.
[image]

You also mentioned previously about trying the query view... Can you give me a tip on how to try that? And any other tips? :) Thanks..

apigott, 2019-08-05:
Was this ever resolved? I'm having the same trackback error using SkySpark...

sjlongland, 2019-08-06:
@apigott Not sure… in this case it seems to be an nHaystack error that's causing the trace-back, and all `pyhaystack` does is throw a `HaystackError` which is basically saying "something went wrong server-side".

About the only thing in common is both nHaystack and SkySpark are (deep down beneath the "Fantom") Java based, hence why you get the `ClassCastException` error (which is a Java exception).  On that basis, even if the stack trace *looks* the same, you're better off filing a new issue as the root cause is unlikely to be identical.

ChristianTremblay, 2019-10-13:
I made many test without being able to reproduce the trouble.
I believe it's related to the server itself.
I will close the issue and please reopen if you find something.

closed 2019-10-13 by ChristianTremblay

bbartling, 2019-10-13:
Any chance you would create a YouTube video on how to setup a haystack
server on N4 secure and pull a trend thru Python? :)

In AX I could use pyhaystack with ease on a non secure station but it's
been a dead end in N4 for me. :(