ChristianTremblay/pyhaystack #77: help with HisReadSeriesOperation

apigott, 2019-08-06:
I've got a session open with skyspark and can pull data via session.about(), etc but HisReadSeriesOperation is giving me issues. My code:

```python
op = pyhaystack.client.ops.his.HisReadSeriesOperation(session, '<point name>', 'today', None, 'dict')
op.go()
op.wait()
```
Always returns op.is_failed = True. The traceback looks like this:

```
Traceback (most recent call last):
  File "pyhay.py", line 64, in <module>
    if (op.result):
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\util\state.py", line 97, in result
    self._result.reraise()
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\util\asyncexc.py", line 28, in reraise
    reraise(*self._exc_info)
  File "C:\Users\USAP677924\api\lib\site-packages\six.py", line 693, in reraise
    raise value
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\client\ops\his.py", line 104, in _on_read
    grid = operation.result
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\util\state.py", line 97, in result
    self._result.reraise()
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\util\asyncexc.py", line 28, in reraise
    reraise(*self._exc_info)
  File "C:\Users\USAP677924\api\lib\site-packages\six.py", line 693, in reraise
    raise value
  File "C:\Users\USAP677924\api\lib\site-packages\pyhaystack\client\ops\grid.py", line 298, in _check_err
    grid.metadata.get('traceback', None))
pyhaystack.exception.HaystackError: sys::CastErr: java.lang.ClassCastException
```
Also I assume that the output of HisReadSeriesOperation is similar to the data series to be written in HisWriteSeriesOperation but the documentation doesn't really elaborate on that...

sjlongland, 2019-08-06:
On 6/8/19 11:17 pm, apigott wrote:
> op = pyhaystack.client.ops.his.HisReadSeriesOperation(session, '<point name>', 'today', None, 'dict')
> op.go()
> op.wait()

Errm yeah, do *NOT* go creating the operation objects yourself.  There
are factory methods on the session object for doing that.  Those factory
methods perform various data normalisation tasks on the inputs before
creating the operation object to manage the actual request.

https://pyhaystack.readthedocs.io/en/latest/his.html#zinc-date-and-time-format
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

ChristianTremblay, 2019-08-09:
One thing possible is to manually create a similar request and check in a browser how the server respond. 

Not always easy but it may help knowing where the bug is : in the server or in the script.

apigott, 2019-08-09:
Thanks guys got it working! @sjlongland was right and I was using HisReadSeriesOperation so did a bit of reading up on factory methods.

ChristianTremblay, 2019-08-09:
@Stuart is awesome ;-)

closed 2019-08-09 by ChristianTremblay