ChristianTremblay/pyhaystack #82: function HayStackSession::_get requests zinc when told to get raw json

criverc, 2019-10-24:
If I do this:

```
from urrlib.parse import quote

session = SkysparkScramHaystackSession('http://address:port', username='someone', password=******, project='someproject', http_args={'tls_verify': False})

request = 'readAll(sometag)'
result = session._get_grid(f'eval?expr={quote(request)}',None,expect_format='json',raw_response=True)
```

Inspecting the request using a network sniffer I see:
```
GET /api/someproject/eval?expr=readAll%28sometag%29 HTTP/1.1
Host: address:port
User-Agent: python-requests/2.21.0
Accept-Encoding: gzip, deflate
Accept: */*
Connection: keep-alive
Authorization: BEARER authToken=sXtUkHhuwVUF0BIZ7cR6xH7tXYRNp7riWfqWSDAmbvGg-2a8
Cookie: skyarc-auth-9080=sXtUkHhuwVUF0BIZ7cR6xH7tXYRNp7riWfqWSDAmbvGg-2a8`
```
The Accept header is wrong it should be: Accept: application/json
It follows that the server returns zinc and not json.

Without the raw_response argument the Accept header is correct. So it seems that the combination of raw_response=True and expect_format='json' does not work as expected.

sjlongland, 2019-10-26:
Could you try pull request #83 and see if that resolves the issue for you?

criverc, 2019-10-29:
I just did that, and merged into develop branch. I run the code snippet that I put in the description of this issue and now it works as expected. I get a binary json array in the body property of the response which I am able to parse using json.loads. Thanks!

sjlongland, 2019-10-29:
No problems at all. :-)

closed 2019-10-29 by sjlongland