ChristianTremblay/pyhaystack #84: Error using his_read()

San42, 2019-11-25:
Hi ,
   We are trying to communicate with Skyspark using the pyhaystack module(python).
The version of of Skyspark and Python we are using are 3.0.22 and 3.7 respectively.
So far we have been successful in connecting with the server and retrieving the site information i.e if we run :
from pyhaystack.client.skyspark import SkysparkScramHaystackSession
session = SkysparkHaystackSession(uri='https://skyspark3.server.org',
                             username='user',
                             password='my_password',
                             project='demo')
session.sites
However, on using the 'his_read' function it is returning a 404 Client error :
Traceback (most recent call last):
 File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/http/sync.py", line 77, in _request
   response.raise_for_status()
 File "/usr/lib/python3/dist-packages/requests/models.py", line 940, in raise_for_status
   raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: GET not allowed for op 'hisRead' for url: https://skyspark3.server.org/api/demo/hisRead?id=%40p%3Ademo%3Ar%3A255873a0-2366039f&range=yesterday
Also attaching the screenshot of the full error message.
Any help on this would be appreciated.
[pyHaystackError.log](https://github.com/ChristianTremblay/pyhaystack/files/3888785/pyHaystackError.log)

sjlongland, 2019-11-26:
Well, unfortunately because you didn't use `\`\`\`` blocks around your pasted error, Github has mangled it so it's a bit difficult to read your bug report.  However, we can see enough to see this is a SkySpark bug.

https://www.project-haystack.org/doc/Rest#getRequests says the above `GET` query is valid.

San42, 2019-11-26:
Just for clarity. 
```Traceback (most recent call last):
	File "/home/pi/. local/lib/python3. 7 /site-packages/pyhaystack/client/http/sync. py ', line 77, in _request
		response.raise_for_status()
	File "/usr/lib/python3/dist-packages/requests/models.py", line 940, in raise_for status
		raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: GET not allowed for op "hisRead" for url: https://skyspark3.server.org/api/demo/hisRead?id=%40p%3Ademo%3Ar%3A255873a0-2366039f&range=yesterday

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
	File "/home/pi/. local/lib/python3. 7 /site-packages/pyhaystack/client/ops/grid. py ', line 288, in _on_response
		response.reraise()
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/util/asyncexc.py" line 29, in reraise
		reraise(~self._exc_info)
	File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
		raise value
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/http/sync.py', line 105, in _request
		e.response.content,
pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 405 Client Error: GET not allowed for op "hisRead" for url: https://skyspark3.server.org/api/demo/hisRead?id=%40p%3Ademo%3Ar%3A255873a0-2366039f&range=yesterday] 405

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
	File "/home/pi/. local/lib/python3. 7 /site-packages/pyhaystack/client/ops/grid. py ', line 288, in _on_response
		response.reraise()
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/util/asyncexc.py" line 29, in reraise
		reraise(~self._exc_info)
	File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
		raise value
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/http/sync.py', line 105, in _request
		e.response.content,
pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 405 Client Error: GET not allowed for op "hisRead" for url: https://skyspark3.server.org/api/demo/hisRead?id=%40p%3Ademo%3Ar%3A255873a0-2366039f&range=yesterday] 405

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
	File "<stdin>", line 1, in <module>
	File "/home/pi/. local/lib/python3. 7 /site-packages/pyhaystack/util/state. py ', line 100, in result
		self._result.reraise()
	File "/home/pi/. local/lib/python3. 7 /site-packages/pyhaystack/util/asyncexc. py ', line 29, in reraise
		reraise(~self._exc_info)
	File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
		raise value
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/ops/grid.py', line 288, in _on_response
		response.reraise()
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/util/asyncexc.py" line 29, in reraise
		reraise(~self._exc_info)
	File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
		raise value
	File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/http/sync.py', line 105, in _request
		e.response.content,
pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 405 Client Error: GET not allowed for op "hisRead" for url: https://skyspark3.server.org/api/demo/hisRead?id=%40p%3Ademo%3Ar%3A255873a0-2366039f&range=yesterday] 405```

So there is no way to use pyhaystack to read(or write) historical values from skyspark ?

dawcurious, 2019-11-26:
I don't remember why his_read didn't work for me too but I'm using pyhaystack to read historical values. Try this
`
from pyhaystack.client.skyspark import SkysparkHaystackSession

session = SkysparkHaystackSession(uri='https://skyspark3.server.org',
    username='user',
    password='my_password',
    project='demo')

# i think stuart suggested get_eval to me a while ago
def get_eval(arg_expr):
    url = 'eval?expr=%s' % arg_expr
    result = session._get_grid(url, callback=lambda *a, **k: None)
    return result

op = get_eval('readAll(point).hisRead(yesterday)')
op.wait()
result = op.result
print(result)
`

San42, 2019-11-26:
Thanks for the replies. I am getting the same error as before even with the code you provided.
```Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/pyhaystack/client/http/sync.py", line 77, in _request
    response.raise_for_status()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: GET not allowed for op 'eval' for url: https://skyspark3.newecology.org/api/demo/eval?expr=readAll(point).hisRead(yesterday)
```
Can you tell me how you are using pyhaystack to read historical values as you had mentioned?

dawcurious, 2019-11-26:
i can't tell why yours is erroring out.
For historical values, I just change the **yesterday** in `readAll(point).hisRead(yesterday)` to let's say `2019-07-21` or `2019-07-21..2019-08-21` for a historical range

sjlongland, 2019-11-26:
On 27/11/19 1:21 am, San42 wrote:
> So there is no way to use pyhaystack to read(or write) historical values from skyspark ?

It should work the way it does now… clearly SkySpark have changed
something that breaks pyhaystack.

I think the onus is on SkySpark to explain their breaking change as it's
not an implementation of Haystack that most of us have regular access to
for testing purposes, so picking up on breaking changes like this is
quite difficult for a small open-source team.

If I was to guess: I'd say they've broken support for GET calls, and so
a patch will be needed to send _all_ requests as POSTs.
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

ChristianTremblay, 2019-11-26:
Technically, if you forget pyhaystack, you should be able to copy paste the request in your browser (once logged in the server) and get the response.
If it doesn’t work on the browser... it wont work in pyhaystack and you must check the configuration of the server.

gmanohar25, 2020-02-10:
Did anyone figure out a solution for this ? Am getting the same error in my deployment.

sjlongland, 2020-02-10:
Not to point out the obvious, but if there had been a solution to this, the solution would be here and the issue would be closed.

In the absence of new information, I'm inclined to believe this is an issue with the SkySpark installation, as the same code works fine with nHaystack and WideSky, and conforms to the documentation for the `hisRead` and `hisWrite` endpoints.  Nothing there says 405 is a valid response.

gmanohar25, 2020-02-11:
@sjlongland  Hi Stuart, Thanks for the response. I talked to a skyspark rep and your guess is correct. This issue is specific to skyspark deployments. From version 3.0.22, they stopped support on GET calls. Code still works for versions before that (i have deployment running perfectly fine for version 3.0.20). I will try to get documentation on this but the rep suggested  to send all request as POST calls.

ChristianTremblay, 2020-02-12:
I've posted this : https://project-haystack.org/forum/topic/787

We'll see...

sjlongland, 2020-02-12:
Well, I'd suggest the Skyspark rep reads [this](https://project-haystack.org/doc/Rest#getRequests).

By responding with HTTP error 405, they have ceased to be Project Haystack compliant.

ChristianTremblay, 2020-02-12:
Ok.... seems you can enable Get feature

> disallowed in version 3.0.20 upwards, but can be enabled using the allowGetWithSideEffects setting.

@gmanohar25 hope this helps... we'll probably have to prepare something eventually as the rules are changing...

closed 2020-02-12 by ChristianTremblay

gmanohar25, 2020-02-13:
@ChristianTremblay Thanks. Am currently using allow Get with side effects to get the data. Rep i talked to send me some documentation i will share below.

The new security features include:
•	Disallow text/plain to be an alias for text/zinc
•	Disallow GET to be called on operations with side effects
•	Cookie attestation on POST requests
All three features are designed to work in concert to protect against a class of attacks known as CSRF.
These features were originally discussed in 3957. If you have custom or third party software communicating with SkySpark over the REST API, then be aware that these security enhancements may cause a breaking change. The ideal solution is to fix the custom/third party code to adhere to these security best practices. However if necessary, then disable via Settings|API.
Each of these features is discussed in more details below.
Disallow Text/Plain
The REST API will now return 415 if Zinc is posted with the "text/plain" MIME type. The fix is to use "text/zinc". If using the Java Haystack client, then please update to the latest version which includes this change.
If using the Accept header to request "text/plain" then 406 will be returned. The fix is to request "text/zinc".
This feature can be disabled via allowTextPlain in Settings|API.
Disallow Get with Side Effects
We have annotated the following REST operators to be "side effect free": about, formats, ops, read. Any other operation is currently considered to have "side effects" which means it might cause state changes such as writes to the database. All operations with side effects are now required to be called with a POST. Calling a side effect operation using GET will now return 405.
The fix is to always make REST calls with POST.
This feature is disabled via allowGetWithSideEffects in Settings|API.
Cookie Attestation
Cookie attestation requires use of a separate token to verify that API requests using cookies can be trusted. We call this second token the attest key - its automatically generated and paired with every authentication session.
This feature only affects cookie authentication. Clients using Haystack authentication with a bearer token are not subject to cookie attestation. When enabled, a POST request relying on cookie authentication must also send the matching attest key via the HTTP header SkyArc-Attest-Key. Otherwise a 400 error is returned. This feature is discussed in more detail in docSkySpark::Auth#attestCookies.
This feature is disabled via attestCookies in Settings|API.

San42, 2020-02-14:
> @ChristianTremblay Thanks. Am currently using allow Get with side effects to get the data. Rep i talked to send me some documentation i will share below.
>
> The new security features include:
> • [...]

This worked for me as well. You have to manually select the "Disallow GET to be called on operations with side effects"  option  for each project I think.

gmanohar25, 2020-02-14:
@San42 Let me know if you face any issues if you have multiple projects running from the same server. My deployments usually have one project per server so it works for me.

clarsen, 2020-04-24:
FWIW, trying to submit hisRead via POST (using the SkysparkScramHaystackSession and sending along both Authorization: BEARER and Cookie headers) still gives an error indicating SkyArc-Attest-Key to be provided.  

```
pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 400 Client Error: SkyArc-Attest-Key required for url: http://XXXXX/api/PROJECT/hisRead?id=%40p%3APROJECT%3Ar%3A2322d76e-b5451e39&range=2020-04-20T13%3A30%3A00-07%3A00+Los_Angeles] 400
```

docSkySpark::Auth#attestCookies that documents this attest key isn't visible to the public.

justindarcy, 2020-07-29:
From reading through this thread along with several others I haven't been able to identify a way to read the data using a GET request without changing the security setting. What are the other options with PyHaystack to read history data from a Skyspark server?  Does this feature still exist in PyHayStack?

sjlongland, 2020-07-29:
To be precise, _no features have been removed from `pyhaystack`_.

SkySpark in their wisdom, decided that they would breach the contract that is "Project Haystack 2.0" and return HTTP error 405 instead of honouring `hisRead` via `GET` requests as was supported in other implementations of Project Haystack at the time.

At the time, doing a `GET` request was a short-hand way of writing a more long-winded `POST` request which had a single grid row.  That worked, `pyhaystack` worked with all three major Project Haystack implementations.

Then SkySpark decided to break that.  Their choice, not ours.  Please don't blame us.

justindarcy, 2020-07-29:
Ahhh, gotcha. So SkySpark decided to no longer be Project Haystack 2.0 compliant which broke the GET requests. 

You say that _At the time, doing a GET request was a short-hand way of writing a more long-winded POST request which had a single grid row._

Is it still possible to retrieve historical trend data from SkySpark using a POST request or any other method for that matter? Or did the change SkySpark made break the possibility to retrieve history data? Forgive me if that is a bad question as I am new to pyhaystack and SkySpark.

sjlongland, 2020-07-29:
> Is it still possible to retrieve historical trend data from SkySpark using a POST request[…]

It is… and I think that's how SkySpark expect all operations to be done going forward.  So it is possible for `pyhaystack` to be changed to support this (and it'll work with other implementations too), just none of us have had the time to do so, and it hasn't been a necessity for nHaystack or WideSky.

justindarcy, 2020-08-19:
> It is… and I think that's how SkySpark expect all operations to be done going forward. So it is possible for `pyhaystack` to be changed to support this (and it'll work with other implementations [...]

Any chance you could point me in the right direction in terms of how to write a POST request to SkySpark? Maybe a link to some resources or example code? I'd be interested to take a shot at getting one working and would be happy to share the code if I am successful.