ChristianTremblay/pyhaystack #85: Authentication issue using domain name

trishogen, 2020-05-18:
Hi there,

Hope you are doing well. I have been trying to use pyhaystack to connect to a SkySpark instance running on a remote server. I am able to do so using the IP address as the URI, but am receiving an error while trying to use our domain name as the URI:  `pyhaystack.client.http.exceptions.HTTPStatusError: [Errno 401 Client Error: Unauthorized for url: https://tipify.brightpowerinc.com/ui] 401` despite using the same credentials and project as I did with the IP address.

I'm able to successfully authenticate using [this node project](https://bitbucket.org/skyfoundry/haystack-auth-node/src/master/) with either the IP or domain name, which leads me to believe the issue may be with pyhaystack but please correct me if I'm wrong.

My company works with Intellastar (one of the SkyFoundary partners) to host our skyspark instance, they were able to provide some context from the SkySpark side, that I thought might be helpful to bring up:

- Authentication requests are being made to `/ui` instead of `/api/projectname/about`
- The requests are being made as GET requests instead of POST requests. I stumbled across [this conversation](https://project-haystack.org/forum/topic/787) So I realize you are already aware of this issue, but was just curious if you had any updates.

Either way, I'm not sure if either of those issues would impact the problem of using a domain name vs the IP but thought I would share.

Thanks in advance,

Trish

sjlongland, 2020-05-20:
I would expect that accessing the site via a hostname shouldn't be a problem, however the fact that the authentication endpoint is different *would* be a problem.

I take it this is a SkySpark fork of some kind?

trishogen, 2020-05-20:
Hmm no, not forked, just imported. I haven't changed the endpoint, `/api/projectname/about` is the suggestion I received for authentication for SkySpark projects.

ChristianTremblay, 2020-05-21:
This is weird...
Any chance we can make some tests ? We don’t have access to a Skyspark server easily.

trishogen, 2020-05-26:
Yeah! I can do some testing and get back to you.

ChristianTremblay, 2020-09-03:
Making test on my side using two instances. It work on both using their URI.

Closing this.

closed 2020-09-03 by ChristianTremblay