ChristianTremblay/pyhaystack #88: SkySpark Authentication

angelinah1, 2020-07-07:
Hi, I enabled the haystack connector on SkySpark. (As I understand it, I do not need to create a specific instance of the haystack connector on SkySpark since that's only needed for SkySpark to retrieve information from another server, although I have tried both ways.) Now, I am trying to connect to a session via pyhaystack by calling the following on JupyterLab: 
```
session = SkysparkHaystackSession(uri='http://server:port/api',
                                  username='user',
                                  password='pw',
                                  project='proj_name',
                                  pint=True)
session.is_logged_in
```
I have tried all combinations of uri's I can think of, including our server's IP address, a localhost address for a local instance of SkySpark, default ports (80, 8080) and port numbers listed on our SkySpark server. However, `session.is_logged_in` always returns `False`, and on the SkySpark end there is also no record of a log-in. 
Could you recommend how to fix this issue? 
Thank you for your help!

ChristianTremblay, 2020-07-07:
What works for me is : 
```
from pyhaystack.client.skyspark import SkysparkScramHaystackSession

session = SkysparkScramHaystackSession(uri='https://demo.skyspark.com',
                                project='demo',
                                username='username',
                                password='password')
```

sjlongland, 2020-07-07:
On 7/7/20 12:38 pm, angelinah1 wrote:
> Hi, I enabled the haystack connector on SkySpark. (As I understand it, I do not need to create a specific instance of the haystack connector on SkySpark since that's only needed for SkySpark to [...]

Did you try issuing a request between these two lines?  pyhaystack
doesn't bother trying to authenticate until it has something to do.

angelinah1, 2020-07-07:
Adjusting these two things fixed it. Thank you both!

closed 2020-07-08 by ChristianTremblay