ChristianTremblay/pyhaystack #89: Writing histories to SkySpark

angelinah1, 2020-07-09:
Hi, I am having trouble writing histories to SkySpark via the `his_write_series `function. I saw [this previous post ](https://project-haystack.org/forum/topic/726)that says the function can take in a dict with keys=datetime.datetime objects and values=data points. I have implemented this such that my `hist_dict `outputs: 
```
hist_dict = {datetime.datetime(2018, 5, 10, 0, 0): 19.26300048828125,
 datetime.datetime(2018, 5, 10, 0, 1): 19.26300048828125, ...} 
```
The data points were read from another point from `start_time `to `end_time`, which are also datetime objects. 
Next, I call: 
```
op = session.his_write_series(samplePointToWrite, hist_dict )
op.wait()
```
However, when I read the history of the samplePointToWrite from `start_time `to `end_time`, an empty MetaSeries is returned. The `his_read_series `function has been working for other points, and there is also no data written on SkySpark. 

Do you have any suggestions on how to proceed? Thank you!

sjlongland, 2020-07-13:
The response to a `hisWrite` is an empty grid (that is, `{"meta": {"ver": "2.0"}, "cols": [{"name": "empty"}], "rows": []}`).

You might try enabling debugging on the `pyhaystack` client and see what it tells you:
```
import logging
import pyhaystack

logger.basicConfig(level=logging.DEBUG, filename='/tmp/skyspark.log')

session = pyhaystack.connect(your, args, here, \
     log=logging.getLogger('pyhaystack'), http_args=dict(debug=true))
```

That will write out `/tmp/skyspark.log` as you make requests, then you can inspect the log to see what went south during the `hisWrite` and the following `hisRead`.

angelinah1, 2020-07-28:
I've only been able to log in via the SkySpark Scram method, should something like this work? 

`session = SkysparkScramHaystackSession(uri=__, username=__, password=__, project=__, log=logging.getLogger('pyhaystack'), http_args=dict(debug=True), pint=True)`

At the moment, I don't see any logs on SkySpark (if that is where it should be writing the logs) other than the login. 

Thank you!

ChristianTremblay, 2020-07-28:
The logs will be on the Python side.

sjlongland, 2020-07-28:
To clarify, I know of no method that would allow a program (written in any language) running on your workstation, to write to the log files on the SkySpark (or any other implementation of Haystack) server.  Doing so would be an egregious security hole.

closed 2020-11-03 by ChristianTremblay