ChristianTremblay/pyhaystack #91: Skyspark needs post requests instead of get requests

ChristianTremblay, 2020-08-20:
Let see how we can figure this out

@justindarcy
@sjlongland I will need your input on that.

From what I remember, all implementations were using the same base (get requests). Question will be do we need to rewrite everything or create something else exclusive for Skyspark...

sjlongland, 2020-08-20:
On 21/8/20 3:04 am, Christian Tremblay wrote:
> From what I remember, all implementations were using the same base (get requests). Question will be do we need to rewrite everything or create something else exclusive for Skyspark... 

I think we can safely do POST requests for all… so maybe an option is to
replace the implementation of `_get_grid` with code that does a POST
with a single-row grid containing what was once query arguments.

Been a long time since I looked at `pyhaystack` code, and work seems to
be moving away from Python as a language in favour of JavaScript.

sjlongland, 2020-08-23:
Okay, so the cookie stuff… that could be tricky as I tried very hard to round-trip cookies set by the server, and now they want us to do the opposite?

ChristianTremblay, 2020-08-23:
@clarsen Submitted where ? 
I missed something ?

sjlongland, 2020-08-23:
https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/http/base.py#L160-L163

Okay, so we need to set `exclude_cookies=True` in requests made to SkySpark.

sjlongland, 2020-08-25:
Okay, so `python-requests` is too clever for its own good I think… it was ignoring me when told to *not* round-trip cookies.

https://github.com/sjlongland/pyhaystack/tree/skyspark-his-read-fix

I actually have converted all `GET` requests to `POST`s, things seem to work.  I don't see it being a problem for the other implementations either.  We should give that a shot and see what happens.

clarsen, 2020-08-25:
> @clarsen Submitted where ?
> I missed something ?
Sorry i only shared 1/2 of the code involved.  https://github.com/ChristianTremblay/pyhaystack/pull/93  has the modifications to strip out cookies.

ChristianTremblay, 2020-08-26:
@sjlongland and I worked really hard and we finally found something that works the way we wanted.

Develop branch have just been updated and all the tests I have made are successful with nhaystack and Skyspark.

For now, only his_read is treated as POST. Next step will be to convert write to POST... but for now, let's celebrate his_read working on skyspark.

@clarsen thanks for your contribution

This will  be merge to master ASAP so PyPI version works for Skyspark. So please give any feedback.

Thanks

sjlongland, 2020-08-26:
@ChristianTremblay  `hisWrite` should be a `POST` request already… as it's often large blocks of data being written, more than what would fit in a `GET` request. :-)

ChristianTremblay, 2020-08-26:
I was thinking about `point_write`

ChristianTremblay, 2020-09-03:
CLosing this as this is working in the develop branch. New release soon

closed 2020-09-03 by ChristianTremblay