ChristianTremblay/pyhaystack #98: Watch_poll does not use argument refresh

Jorricks, 2020-10-01:
Hello,

We are trying to set up a watch for a skyspark installation. 
I was trying to get `watch_poll` to work, however, I noticed that it always returns me all the points whether I have refresh on True or False.
I did some digging and found out that it has to do with the following piece of code:
https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/session.py#L510
Here it basically forgets about the refresh parameter.

I am unfamiliar with the haystack documentation and on how a watch works internally and therefore I am not aware whether this is a quick fix.

sjlongland, 2020-10-01:
Yeah, that'd be a whoopsie on my part… might've _meant_ to put those two lines in, but didn't.

Jorricks, 2020-10-05:
Hi @sjlongland,

Thank you for adding this :)!

closed 2020-11-03 by ChristianTremblay