ChristianTremblay/pyhaystack #99: Implementation of ability to add new entities?

mschrader15, 2020-10-01:
Hey All,

Ran into this:
https://github.com/ChristianTremblay/pyhaystack/blob/3c3508212c0f5a4e5f254fed4d169da91f03242d/pyhaystack/client/entity/model.py#L61

Is it possible to add new entitles to SkySpark with pyhaystack? Looking to automate the process of adding equipment and sensor points. 

Thanks!

sjlongland, 2020-10-01:
If you can point us to documentation on how to do that, sure.  We'll look at adding it.

At the moment, the only way I know of in nHaystack to edit the asset model is via the workbench, and I have no idea how entities get into SkySpark.  The only implementation that I know of which supports this is WideSky.

sjlongland, 2020-10-01:
Related issue: https://github.com/ChristianTremblay/pyhaystack/issues/74

mschrader15, 2020-10-02:
@sjlongland You are probably right. I will reach out to SkySpark and see if it is even possible per #74 

I'll close this issue for now

mschrader15, 2021-01-07:
Follow up to this, I was able to add new equipment points to our skyspark server using:

```
    def _create_equipment_grid(self, equip_name, markers):
        grid = hszinc.Grid()
        grid.metadata['commit'] = 'add'
        # grid.metadata['projName'] = 'Springfield'
        grid.column['dis'] = {}
        grid.column["equip"] = {}
        grid.column["siteRef"] = {}
        grid.column["navName"] = {}
        #grid.column['disMacro'] = {}
        g = {'dis': equip_name, 'navName': equip_name, 'equip': hszinc.MARKER, "siteRef": self._site.id}
        for marker in markers:
            grid.column[marker] = {}
            g[marker] = hszinc.MARKER
        grid.append(g)
        return grid
```

and then using PyHaystack's protected _post_grid method:

```
    def _post_grid(self, g):
        r = self.session._post_grid(grid=g, callback=None, uri='commit')
        return r
```

sjlongland, 2021-01-10:
This is an interesting development indeed.

To achieve parity with what WideSky permits, we'd need the ability to:
* add new entities (`commit` endpoint looks to be the one that does this)
* mutate entities (maybe doable via `commit` as well) by:
** adding new tags
** changing the values of existing tags
** deleting tags
* delete entities

If we can do those things, say via `curl`, we can most certainly implement it in `pyhaystack`.  The "CRUD" operations in WideSky would be a good starting point for implementing this.

mschrader15, 2021-01-10:
I have had success mutating existing entities in SkySpark using: 
`grid.metadata['commit'] = 'update'` **instead of** `grid.metadata['commit'] = 'add'`
I have used this to add new tags, though haven't tried to overwrite or remove existing tags. 

We actually just ran into a use case where it would've been convenient to delete entities using `pyhaystack`, but the low # of entities made it more efficient to just delete them using SkySparks UI

ChristianTremblay, 2021-01-14:
I like that, it adds a nice scripting possibility to pyhaystack.