ChristianTremblay/pyhaystack #101: Issue with NaN values

Jorricks, 2020-10-05:
Hello all,

First of all, thank you very much for this awesome open source software.
I am very happy with it's state.

I did notice a minor bug though.

Easy to reproduce:
```
<Grid>
	Version: 3.0
	Metadata: MetadataObject{'id'=Ref('p:some:r:sensor', 'Light intensity  []', True)}
	Columns:
		ts
		val
	Row    0:
	ts=datetime.datetime(2020, 10, 5, 17, 0, tzinfo=<DstTzInfo 'Europe/Amsterdam' CEST+2:00:00 DST>)
	val=BasicQuantity(9015.6689453125, 'lm')
	Row    1:
	ts=datetime.datetime(2020, 10, 5, 17, 8, tzinfo=<DstTzInfo 'Europe/Amsterdam' CEST+2:00:00 DST>)
	val=NA
</Grid>
```

In the current code in _pyhaystack/clients/ops/his.py_ the NA value is not properly processed.
It will lead to the following error.
```
Traceback (most recent call last):
  File "sky_spark.py", line 160, in backfill_one_point
    data: MetaSeries = query.result.to_dict()
  File "venv/lib/python3.8/site-packages/pyhaystack/util/state.py", line 100, in result
    self._result.reraise()
  File "venv/lib/python3.8/site-packages/pyhaystack/util/asyncexc.py", line 29, in reraise
    reraise(*self._exc_info)
  File "venv/lib/python3.8/site-packages/six.py", line 703, in reraise
    raise value
  File "venv/lib/python3.8/site-packages/pyhaystack/client/ops/his.py", line 157, in _on_read
    meta_serie = MetaSeries(data=values, index=index)
  File "venv/lib/python3.8/site-packages/pandas/core/series.py", line 316, in __init__
    raise ValueError(
ValueError: Length of passed values is 1, index implies 2.
```

My proposal is to fix this with a small addition after line 146, where the code would become:

```python
except AttributeError:
    if isinstance(each, float):
        values.append(each)
    else:
        values.append(np.nan)
    continue
```

And of course in the top, next to pandas, `import numpy as np`.
As we panda requires numpy to be installed, I think it should be fine to import numpy.

Is this considered a reasonable solution?

ChristianTremblay, 2020-11-03:
This seems pretty reasonable yes.
Thanks for pointing that out

ChristianTremblay, 2020-11-03:
Please test and if it works, we'll close

Jorricks, 2020-11-03:
Tested. Worked well for me. Thanks!

closed 2020-11-04 by ChristianTremblay