ChristianTremblay/pyhaystack #102: Retrieving point write status was broken in 3.0?

jdechalendar, 2020-10-29:
Previous behavior for the `point_write` method [here](https://github.com/ChristianTremblay/pyhaystack/blob/ea6177c8d57227331df60dde4aab8fa0daec0118/pyhaystack/client/session.py#L234-L253) was to retrieve the write status of the point if only the `point` argument was supplied (causing all other arguments to be `None`).

Since the 3.0 release, more specifically a78f5889997217e8e5afe16ca252e063b9025207, if the `who` kwarg is not provided, `self._username` is used instead.

This seems to break the previous behavior of calling `point_write(point)`, as `who` is no longer `None`.

Can you provide guidance for how to retrieve point write status in 3.0? Is this behavior intended?

ChristianTremblay, 2020-11-03:
What if you explicitly pass `who=None` ?

ChristianTremblay, 2020-11-03:
Forgot to ask, using this in skyspark ? nhaystack or widesky ?

ChristianTremblay, 2020-11-03:
> What if you explicitly pass `who=None` ?

Nevermind... it'll be overridden...

ChristianTremblay, 2020-11-03:
Please, try the develop branch where I modified the behaviour so who will be reset to None in the case where level is None.

jdechalendar, 2020-11-23:
Sorry for not replying earlier. Just saw the change [here](https://github.com/ChristianTremblay/pyhaystack/blob/develop/pyhaystack/client/session.py#L525-L530)
- Does it make sense to change the error message to remove the requirement for `who` to be `None` as the value for `who` will not be considered if `level` is `None`.
- In the first part of the if statement, you set `who` to be `None`, but that does seem to be used - is that line needed?

With these suggestions, the `on_point_write` method would become:
```python
def _on_point_write(self, point, level, val, who, duration, callback, **kwargs):
        args = {"id": self._obj_to_ref(point)}
        if level is None:  # Ignore value for who as it may be username
            if (val is not None) or (duration is not None):
                raise ValueError(
                    "If level is None, val and duration must be None too."
                )
        else:
            args.update({"level": level, "val": val})
            if who is not None:
                args["who"] = who
            if duration is not None:
                args["duration"] = duration
        return self._get_grid("pointWrite", callback, args=args, **kwargs)
```

It may also be worth updating the docstring of the `point_write` method to reflect the change: If level is None, the val and duration parameters should be None too, the write status of the point is retrieved.

ChristianTremblay, 2020-12-02:
You are right. who is not used. We can remove that line.
Merci !
Regarding the docstring, I'm heading toward this, hoping it will be clearer.

        """
        point is either the ID of the writeable point entity, or an instance of
        the writeable point entity to retrieve the write status of or write a
        value to.

        To retrieve the write status of the point, value, duration and level are required to be None.
        To write to a point, value and level are required and duration is optional.
        """

Function will become

    def _on_point_write(self, point, level, val, who, duration, callback, **kwargs):
        args = {"id": self._obj_to_ref(point)}
        if level is None:
            if (val is not None) or (duration is not None):
                raise ValueError(
                    "You tried writing a value without specifying a level (or specified a duration without providing a level neither a value. Please provide required level. If trying to retrieve write status of the point, provide None as value, duration and level"
                )
        else:
            args.update({"level": level, "val": val})
            if who is not None:
                args["who"] = who
            if duration is not None:
                args["duration"] = duration
        return self._get_grid("pointWrite", callback, args=args, **kwargs)

closed 2021-02-26 by ChristianTremblay