ChristianTremblay/pyhaystack #109: LDAP login on SkySpark 3.0.28

gidnoot, 2021-05-21:
When trying to login to my Active Directory account on SkySpark, I get an AuthenticationProblem exception. But when I try a non LDAP account, it just works fine. I am sure that my password is correct. Does someone know what's going on here?

Code:
```python
import pyhaystack

session = pyhaystack.connect(implementation='skyspark',
                             uri='https://<address>.nl',
                             username='g.noya',
                             password='<password>',
                             project='master',
                             pint=True)

session.get_eval('read(func)').result
```

output:
    raise AuthenticationProblem()
pyhaystack.exception.AuthenticationProblem

Maybe good to know is that I first couldn't login because my username contained a dot in it. With SkySpark 3.0.28 this issue is solved but maybe not for Pyhaystack

sjlongland, 2021-05-21:
LDAP or something else really should be SkySpark's problem, not the problem of the Haystack client.

Likewise usernames with dots: `pyhaystack` doesn't do something special when it encounters a dot in a username.  Unless there's some funny business going on with SCRAM auth, I'd suggest there's a bug in Skyspark since usernames with dots in them work fine on other implementations like WideSky (which uses email addresses as user names).

gidnoot, 2021-05-23:
Thank you so much for your response. I also posted it on their forum so then they have to fix it :)

gidnoot, 2021-05-25:
Brian Frank (Senior Software Developer of SkySpark) said this:
To login with LDAP requires that the password be transported in plaintext to the server which we do with an non-standard "x-plaintext" auth scheme. So I'm pretty sure the Python client doesn't support it since its not part of the standard. You can ask the guys who support it if they want to add it and I'll work with them

So could this be added?

ChristianTremblay, 2021-05-31:
This could..... but "plain-text" ???????
why bother with a password then ?

sjlongland, 2021-06-03:
Seems they'd have been better off just using [RFC-2617](https://datatracker.ietf.org/doc/html/rfc2617) since sending the password in clear text is clearly not a concern.  Or use [OAuth2](https://datatracker.ietf.org/doc/html/rfc6749) and at most send the password _once_.

How does the Haystack client know the server is using this contrived scheme?

gidnoot, 2022-08-24:
They've implemented the PLAINTEXT auth scheme. This is used on users that use LDAP. This works fine when I send a request without PyHaystack but I haven't figured out how to create a session with PyHaystack using PLAINTEXT auth. Is there already a way to do this or does it have to be implemented into PyHaystack first?

For more information on the PLAINTEXT auth scheme, read this page: https://project-haystack.org/doc/docHaystack/Auth#plaintext