ChristianTremblay/pyhaystack #110: Axon eval(hisClear) requires POST, not GET with skyspark

clarsen, 2021-06-10:
```python
session.get_eval( 'hisClear(parseRef("p:dataCenter50B:r:282c63c3-f1e7d0c0"), 2021)' )
```

results in
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/pyhaystack-3.0.0-py3.8.egg/pyhaystack/client/http/sync.py", line 78, in _request
  File "/usr/local/lib/python3.8/site-packages/requests-2.25.1-py3.8.egg/requests/models.py", line 943, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 405 Client Error: GET not allowed for op 'eval' for url: https://skyspark.lbl.gov/api/dataCenter50B/eval?expr=hisClear(parseRef(%22p:dataCenter50B:r:282c63c3-8655c4e7%22),%202021)
```
with fix of using _post_grid:
```python
class EvalOpsMixin(object):
    def post_eval(self, arg_expr):
        url = "eval"
        eval_grid = hszinc.Grid()
        eval_grid.column["expr"] = {}
        eval_grid.append({"expr": arg_expr})
        return self._post_grid(url, eval_grid, callback=lambda *a, **k: None)
```

results in
```
<Grid>
        Version: 3.0
        Columns:
                id
                dis
                numRemoved
        Row    0:
        id=Ref('p:dataCenter50B:r:282c63c3-f1e7d0c0', '91 onicon  Return Temperature', True)
        dis='91 onicon  Return Temperature'
        numRemoved=1918.0
</Grid>
```

if this looks acceptable, i can submit a PR...

ChristianTremblay, 2021-06-11:
Can you try develop branch ? We've made a post_eval version few months ago

jdkaire, 2022-03-28:
@ChristianTremblay I see the code is still in the develop branch.    @clarsen can you submit the PR? 
Thanks!

Rinke-beep, 2024-10-03:
Are there any updates on this issue? I am still having the above mentioned 405 errors using pyhaystack 3.0.0?