ChristianTremblay/pyhaystack #113: Pint error?

brandonfwl, 2023-01-09:
First time attempting to use pyhaystack.  When connecting to an n4 instance with example usage pyhastack.connect I get an error from pint.

`pint.delegates.txt_defparser.common.DefinitionSyntaxError: Cannot define '%' (UnitDefinition): is not a valid unit name (must follow Python identifier rules)
    1,0-1,16 % = [] = percent`

sjlongland, 2023-01-09:
This is actually an issue with a dependency: `hszinc`.

https://github.com/widesky/hszinc/issues/38

brandonfwl, 2023-01-09:
Installed pint==0.19.2 and that seemed to resolve the issue.  Any plans to resolve the issue?

sjlongland, 2023-01-09:
On Mon, 09 Jan 2023 15:32:39 -0800
Brandon Fowler ***@***.***> wrote:

> Installed pint==0.19.2 and that seemed to resolve the issue.  Any
> plans to resolve the issue?

When I get a moment at work to work on `hszinc`, yes I do plan to
investigate the issue.  Sadly, it's not ${MEGACORP} with a workforce of
millions to work on this stuff, it's individuals with back-logs tall
enough to require warning lights to warn flying aircraft at night.