ChristianTremblay/pyhaystack #114: Possible error while fetching data from a Niagara server hosted on HTTPS

krishabh-95, 2023-04-12:
Hi,

I am trying to connect to a few Niagara 4 instances using pyhaystack. When I try to connect to the instances hosted on http, it works fine but when I try to connect to a Niagara4 instance hosted on https, it throws an error. I am using the following code to establish a connection:

**from pyhaystack.client.niagara import Niagara4HaystackSession**
**session = Niagara4HaystackSession(uri=niagara_host, username=niagara_user, password=niagara_password)**

Should I do something different to connect to a https instance of Niagara4? Just so you know, haystack service has been enabled in this instance since I am able to fetch the data when I execute the URL <niagara-host>/haystack/about in the browser. Thanks in advance.

sjlongland, 2023-04-12:
You mention it "throws an error" but don't say which one… I'm guessing possibly a certificate error?

There are a couple of ways around this:

1. disabling TLS verification altogether (not the wisest idea, but can work): specify `http_args=dict(tls_verify=False)` in the session constructor arguments (along side the `uri`)
2. obtain a copy of the certificate from the remote server, and specify the path to that file; http_args=dict(tls_verify="/path/to/remote.crt") in the session constructor arguments.

krishabh-95, 2023-04-12:
Sorry I forgot to mention the exact error. Yes, it was a certificate error indeed. It works fine now after disabling TLS verification. Thanks a lot for your help.

I'll try to get a copy of the certificate from the remote server and use it for TLS verification.

closed 2023-06-03 by krishabh-95