ChristianTremblay/pyhaystack #115: Parsing issue evalOrReadAll(\"ruleSparkSwivel([@nav:equip.all], toSpan(\\\"yesterday\\\"), [], {})\")

rafayehmed, 2023-07-11:
We are using pyhastack to get data from skyspark.
```
op = session._post_grid("evalAll",
                            eval_grid,
                            callback=lambda *a, **k: None)
evalOrReadAll(\"ruleSparkSwivel([@nav:equip.all], toSpan(\\\"yesterday\\\"), [], {})\")
```

we are getting data in this form lets say.

ver:"3.0" viz:"genTable" presentation:<<
ver:"3.0" disableSel useGrouping hideRowInfo
col,row,color,padding,halign,nav,viz,info,format
0,,,,,"targets",,,
2,,,,,,,,
3,":header",,"0",,,"runtimeAxisCell",,
3,,,"0",,,,,
4,,,,,,,,
5,,,,,,,,
[2,3,4,5,1],[8,13,18,23,42],"#9b59b6",,,,,,
[2,3,4,5,1],[45],"#e67e22",,,,,,
"ruleRef_dis",,,,,,,"fandoc",
"targetRef_equip_id",,,,,"_swivelRefList",,"special","_swivelRefList"

>> navPath:[@nav:equip.all "All Sites"] periodsInterval:1min ruleInfo:<<
ver:"3.0"
color,dis,help
"#9b59b6","m61HighCurrentTHD","1. Determine if high current THD is affecting voltage THD and the rest of your system.  2. Determine if adding harmonic filters will benefit your power system.\n\nHigh current THD can apply breaking force to motors, cause windings to overheat, and shafts to flute or pit."
"#e67e22","CHW pump differential pressure < Diff SP - 1 IWC for more than 2 hours.","Identify why pump differential pressure is not resetting."

>> hisStart:2023-07-10T00:00:00Z Rel hisEnd:2023-07-11T00:00:00Z Rel
targetRef_site select:"targetRef.site" navDiveDown dis:"Site",ruleRef_dis select:"ruleRef.dis" dis:"Rule",dur fold:"sum" select:"dur" viz:"barCell" dis:"Duration",periods fold:"periodUnion" select:"periods" viz:"runtimeCell" dis:"Periods",targetRef_equip_id fold:"list" select:"targetRef.equip.id" viz:"text" dis:"Equips",targetRef_site_id fold:"list" select:"targetRef.site.id" hidden dis:"targetRef.site.id"
@nav:equip.site.aWQ6QHA6dGZzU21hcnRCdWlsZGluZ3M6cjoyYjk0NDI4NS1hZDhmODRhYw "Eugene, 29851 Willow Creek Road - Bldg J","m24VFDCurrentIncrease",5.75h,"AABz",[[@p:tfsSmartBuildings:r:2bc876e9-7676d349 "Chiller- (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-e07d527b "Vacuum Pump-2 (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-94fa637e "Vacuum Pump-1 (AcuRev)",1]],[[@p:tfsSmartBuildings:r:2b944285-ad8f84ac "Eugene, 29851 Willow Creek Road - Bldg J",3]]
@nav:equip.site.aWQ6QHA6dGZzU21hcnRCdWlsZGluZ3M6cjoyYjk0NDI4NS1hZDhmODRhYw "Eugene, 29851 Willow Creek Road - Bldg J","m36KWhigher",6h,"AAB4",[[@p:tfsSmartBuildings:r:2bc876e9-7676d349 "Chiller- (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-e07d527b "Vacuum Pump-2 (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-94fa637e "Vacuum Pump-1 (AcuRev)",1]],[[@p:tfsSmartBuildings:r:2b944285-ad8f84ac "Eugene, 29851 Willow Creek Road - Bldg J",3]]
@nav:equip.site.aWQ6QHA6dGZzU21hcnRCdWlsZGluZ3M6cjoyYjk0NDI4NS1hZDhmODRhYw "Eugene, 29851 Willow Creek Road - Bldg J","m61HighCurrentTHD",62.77788833333334h,"AATo",[[@p:tfsSmartBuildings:r:2bc876e9-7676d349 "Chiller- (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-e07d527b "Vacuum Pump-2 (AcuRev)",1],[@p:tfsSmartBuildings:r:2bc87705-94fa637e "Vacuum Pump-1 (AcuRev)",1]],[[@p:tfsSmartBuildings:r:2b944285-ad8f84ac "Eugene, 29851 Willow Creek Road - Bldg J",3]]
"p:tfsSmartBuildings:r:2b944285-ad8f84ac","Find Missing Data on Acurev points",132h,"AAKU",[[@p:tfsAWSHub:r:2c14d8fd-e7404e70 "29851 Willow Creek - Bldg J AHUs (AcuRev Bacnet)",12]],
"p:thermofisher:r:27378e59-0378917c","Find Missing Data",8h,"A8Hg",[[@p:tfsAWSHub:r:28fc86ac-f16b9e17 "GasMeter - 62228360 (AWShub)",1]],
"p:thermofisher:r:27378e59-890391b7","Find Missing Data",8h,"A8Hg",[[@p:tfsAWSHub:r:28fc86ac-19b7ecbd "GasMeter - 47267385 (AWShub)",1]],
"p:thermofisher:r:2737ac20-2616a491","Find Missing Data on Acurev points",63h,"AAIc",[[@p:tfsAWSHub:r:2aabfb46-c981d19e "ElecMeter - #6700265",2],[@p:tfsAWSHub:r:2aabfb46-4b46faf6 "ElecMeter - #6700262",2],[@p:tfsAWSHub:r:2aabfa81-84c02922 "ElecMeter - #6583881",2],[@p:tfsAWSHub:r:2aabfa81-65b84a49 "GasMeter - #1418931",1]],
"p:thermofisher:r:2737acdc-d1ad5e27","Find Missing Data on Acurev points",18h,"AAIc",[[@p:tfsAWSHub:r:2aaa93a8-0d7073c2 "ElecMeter - #590512",2]],
"p:thermofisher:r:2737adba-9ef51ee1","Find Missing Data on Acurev points",18h,"AAIc",[[@p:tfsAWSHub:r:2aaa93ff-7f207bfc "ElecMeter - #590553",2]]

But we are getting this error

Expected '\n', found ':'  (at char 37), (line:1, col:38)
Expected '\n', found ':' (line:1, col:38)
Expected '\n', found ':'  (at char 37), (line:1, col:38)
Expected '\n', found ':' (line:1, col:38)
Expected '\n', found ':'  (at char 37), (line:1, col:38)
Expected '\n', found ':' (line:1, col:38)
Traceback (most recent call last):

I want to convert this data into json form. Can any one please assist.

sjlongland, 2023-07-15:
It's a bit difficult to discern what is actually getting returned.  Github Issues use Markdown formatting, and I think some of what you've posted is getting interpreted as Markdown (and a ping to fellow Github user @p who is now probably wondering what this barrage of mentions is about).

It's worth noting anything starting with an underscore (`_`) is protected interface, and not a guaranteed stable API.  We might be able to add an `eval_all` method to the SkySpark interface if you can provide some detail of how it works so we can better support it, however I'd strongly recommend against any production scripts directly calling `_post_grid` (or its related `_get_grid`).

The other thing you can try is when creating your session; specify `grid_format=hszinc.MODE_JSON`: as it looks like you've requested and got back ZINC encoding and a highly complicated ZINC grid to boot.  `pyhaystack` uses `hszinc` for ZINC parsing/encoding, so if there's an issue parsing the grid, it'll be a bug in `hszinc` not `pyhaystack`.

ChristianTremblay, 2023-07-15:
Agree with Stuart... use JSON