ChristianTremblay/pyhaystack #116: Use query function for Naigara4 nHaystack Server

MGodK-NPK, 2023-12-22:
# Problem
## Example Code:

### Code 1
- Prompt
```
from pyhaystack.client.niagara import Niagara4HaystackSession
from pyhaystack.client.http.auth import BasicAuthenticationCredentials
import hszinc

URI  = "server endpoint"
USERNAME  = "username"
PASSWORD  = "password"

authenticator = BasicAuthenticationCredentials(
                                                    USERNAME, 
                                                    PASSWORD
                                                )

json = hszinc.MODE_JSON

session = Niagara4HaystackSession(  
                                            uri = URI,
                                            username = authenticator.username,
                                            password = authenticator.password,
                                            pint = True,
                                            grid_format= json
                                        )

session
```
- Output
```
<pyhaystack.client.niagara.Niagara4HaystackSession at 0x12d8aa110>
```

### Code 2
- Prompt
```
session.sites
```

- Output
```
HaystackError: java.lang.ClassCastException: org.projecthaystack.HStr cannot be cast to org.projecthaystack.HRef
```
[image]

### Code 3
- Prompt
```
floor_2 = session.get_entity(
                                ids= "S.Floor_2"
                            )
floor_2.wait()
floor_2.result
```

- Output
```
<@S.Floor_2: {area=0.0, axSlotPath='slot:/HS_Ontology/Floor_2', axType='nhaystack:HSite', dis='Floor_2', geoAddr='', geoCity='', geoCoord......, navName='Floor_2', site, tz='Bangkok'}>
```

## Question
- I can get entity by get_entity function but can not find_entity

Picture for tagging and ontology
[image]

MGodK-NPK, 2023-12-22:
But if i test on another JACE that have ~900+ points and ~250 equips and 3 sites.

It is well everything I can use every function in pyhaystack.

In this problem case, I test on real site that have ~20 in each modbus device(equip).

I don't know where is error from?

sjlongland, 2023-12-22:
Okay, so `java.lang.ClassCastException` is a server-side exception… caused when you try to cast an object to the wrong sub-class.  In this case, it seems to be expecting a `HRef` and got a `HStr` instead.

https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/session.py#L405-L420 is pretty simple.  So it's doing a `find_entity` call… unwisely it's not `wait()`-ing on the result, so this property will only work with the synchronous client (which is default)… but effectively in Haystack-land, it's doing a `read?filter="site"`.

`filter=` in a `read` call is supposed to be a `HStr`, and from what I see in the code, it is a `HStr`.

If the JACE is expecting something else there, something is seriously wrong.  I suspect a bug or misconfiguration in the server.

I'd do the following:

```python
from pyhaystack.exception import HaystackError

try:
    print(session.sites)  # will most likely throw the exception as seen before
except HaystackError as e:
    print(e)
    print(e.traceback)
```

That'll show you the traceback from the server-side, which you can take up with nHaystack people if needed.

MGodK-NPK, 2023-12-22:
[image]

I use your script for testing an error

sjlongland, 2023-12-22:
Right, so server is suppressing the stack trace (probably a good thing, you don't want to show it to an attacker, nHaystack used to reveal it by default)… but you'll have to look in Niagara Workbench (probably the platform manager if I recall correctly) to see what's going on server-side.