ChristianTremblay/pyhaystack #118: pyhaystack.util.state.NotReadyError

showfuture, 2024-03-14:
python: 3.12.1
pyhaystack:       3.0.0
[image]

```
    class OpticManager:

        def __init__(self) -> None:
            self.url = f'{SIEMENS_HOST}'

            self.client = pyhaystack.connect(implementation='skyspark',
                                             uri=self.url,
                                             username=SIEMENS_USER_NAME,
                                             password=SIEMENS_PASSWORD,
                                             project=SIEMENS_PROJECT_NAME,
                                             pint=True)

        def get_about(self):
            op = self.client.about()
            op.wait(3)
            about = op.result()
            print(about)

    optic = OpticManager()
    optic.get_about()
```

When I run this code, it throws an error ：
```
File "/root/.local/share/virtualenvs/siemens-edge-i-9CWoel/lib/python3.12/site-packages/pyhaystack/util/state.py", line 97, in result
    raise NotReadyError()
pyhaystack.util.state.NotReadyError
```

Why is this happening? Is there something wrong with my code logic?

sjlongland, 2024-03-14:
Means the request you issued had not been completed by the time you asked for the result.  While 3 seconds should be enough time, I don't know enough about the environment.

Configuring the `logging` module might help unlock more clues too:

```
import logging

logging.basicConfig(level=logging.DEBUG)
```

showfuture, 2024-03-14:
If I don't set the wait for 3 seconds, it will remain blocked indefinitely.  logging also  not work.

sjlongland, 2024-03-14:
Then there is something that is preventing the request from being processed _at all_.  `.wait(3)` means wait for up to 3 seconds for it to finish.

The result will be available when it's fulfilled by the server and library, not before.  So if after 3 seconds, it still isn't done, and you ask for the result, it'll error out the way it did above.

Why didn't it complete?  Who knows?  The above logs don't tell us.  I refer you to https://github.com/ChristianTremblay/pyhaystack/issues/118#issuecomment-1996314951 for advice on how you can troubleshoot it earlier.

I can't say I've ever tested on FINStack, so no telling whether it'd work or not.

showfuture, 2024-03-14:
ok ,  Is there something wrong with my code logic?

showfuture, 2024-03-14:
```
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 172.16.50.199:8080
DEBUG:urllib3.connectionpool:http://172.16.50.199:8080 "GET /user/login HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 172.16.50.199:8080
DEBUG:urllib3.connectionpool:http://172.16.50.199:8080 "GET /ui HTTP/1.1" 404 241
```
note this error

sjlongland, 2024-03-14:
Right, sounds like the SCRAM auth is not quite working properly and the Haystack thinks we're a web browser, thus sends us to the UI page to interactively authenticate.

Looks like it's failing right here:
https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/ops/vendor/skyspark_scram.py#L134

Not sure why it gets stuck at that point, authentication should have failed, but in any case, it appears either this implementation is not compatible with Skyspark, or they've changed their protocols yet again.

showfuture, 2024-03-15:
```
import logging

    logging.basicConfig(level=logging.DEBUG, filename='siemens.log')

    session = SkysparkScramHaystackSession(
        uri=SIEMENS_HOST,
        username=SIEMENS_USER_NAME,
        password=SIEMENS_PASSWORD,
        project=SIEMENS_PROJECT_NAME
    )
    print(f"Session state: {session.is_logged_in}")

    au = session.authenticate()
    print('au', au)
```

package:   fysom/__init__.py

the code run ,but throw this error:
[image]

```
import logging

    logging.basicConfig(level=logging.DEBUG, filename='siemens.log')

    session = SkysparkScramHaystackSession(
        uri=SIEMENS_HOST,
        username=SIEMENS_USER_NAME,
        password=SIEMENS_PASSWORD,
        project=SIEMENS_PROJECT_NAME
    )
    print(f"Session state: {session.is_logged_in}")

    # au = session.authenticate()
    # print('au', au)

    print(session.find_entity("site"))
```

package:   fysom/__init__.py

the code run ,but throw this error:

[image]

And the log is 
```
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): 39.97.226.236:8080
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): 39.97.226.236:8080
```

sjlongland, 2024-03-17:
`fysom` is a state-machine library, and yes, we define an "exception" state which gets passed through to the `Operation` class for throwing an exception.

As I say, I don't know FIN|Stack.  This is not a Haystack implementation I've worked on, nor have access to.  It's a big assumption to presume compatibility with Skyspark.  It is also possible that there is a bug in Skyspark support code too.  It doesn't get a lot of "love".

I said as much in https://github.com/ChristianTremblay/pyhaystack/issues/118#issuecomment-1997034253.

HTTP vs HTTPS: this will depend on what your server's configuration is.  Whatever your web browser uses if you use a web browser to interact with FIN|Stack, you should use the same protocol in `pyhaystack` unless there's something specific about your network that suggests otherwise.