ChristianTremblay/pyhaystack #121: read vs watch_poll

aalvarez89, 2025-06-05:
Hello, I needed to inquire regarding the functionality of a watch object.
My objective was basically subscribing to a data point and getting a live feed of that data point's value.
I'm not sure what am I doing wrong but after creating the watch object, I have to constantly ping again the server with the watch_poll() function, so how is this different than just calling .read() on that same data point?
Is there actually a way to get a live feed with one call?
Thanks in advance

aalvarez89, 2025-06-05:
@sjlongland  Understood, that makes a lot of sense, then what is the callback on .watch_poll() for? or how do you properly use it?
Thanks!

sjlongland, 2025-06-09:
Sounds like a bug to be honest, would you have a full backtrace so we can see where it is happening?

aalvarez89, 2025-06-10:
```
python hs_live.py

<PostGridOperation failed>

Traceback (most recent call last):
  File "~/hs_live.py", line 39, in <module>
    call_watch_poll(ws_id, True, poll_cb)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^

  File "~/hs_live.py", line 35, in call_watch_poll
    session.watch_poll(watch_id, refresh, callback)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "~/venv/lib/python3.13/site-packages/pyhaystack/client/session.py", line 232, in watch_poll
    return self._on_watch_poll(watch=watch, refresh=refresh, callback=callback)
           ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "~/venv/lib/python3.13/site-packages/pyhaystack/client/session.py", line 515, in _on_watch_poll
    watch = watch.id
            ^^^^^^^^

AttributeError: 'PostGridOperation' object has no attribute 'id'

```
@sjlongland

sjlongland, 2025-06-10:
Okay, so it's been quite some time since I've looked at this code (my workplace used to use Python a lot, but in 2016 pivoted away from that to JavaScript).

Two things stick out:
* `PostGridOperation` is a state machine that handles encoding, submission, retrieval and decoding of HTTP POST requests and responses.  It itself isn't meant to be used directly: you're supposed to retrieve the result of the operation once it's done; either `await` it if it's Python3, or call `.wait()` to synchronously block on it, then read the `.result` property.
* The `watch_id` is not the whole grid returned by `watchSub`, but rather, the `watchId` field given in the metadata.

The `watch_sub`, `watch_poll` and `watch_unsub` are low-level functions that merely encode the arguments into a grid to be submitted to the Haystack server, and return the result as a Grid object.  There isn't a higher-level abstraction at this stage as there hasn't been a requirement for one so far (real-time doesn't get a lot of love).

So if you call `watch_sub`, and the server responds with a grid like this:

```
ver:"3.0" watchId:"w-2bfe7e77-8a7c55ab" lease:1min
id,cur,curStatus,curVal,equipRef,kind,point,unit
@2b80f96a-9dcc2202 "RTU-1 ReturnTemp",M,"ok",68.2°F,@2b80f96a-ee1d1270 "RTU-1","Number",M,"°F"
@2b80f96a-7b16eb30 "RTU-1 ZoneTemp",M,"ok",74.1°F,@2b80f96a-ee1d1270 "RTU-1","Number",M,"°F"
```

The thing you pass to `watch_poll` is the string `"w-2bfe7e77-8a7c55ab"`:

```python
ws_id = get_watch_poll_grid() # I can't see the code that sets `ws_id`
watchsub_res = await ws_id

watchpoll_res = await session.watch_poll(ws_id.result.metadata["watchId"], refresh, callback)
```

That would give you a grid object in `watchpoll_res`.  Let's assume the server gave us this:

```
ver:"3.0" watchId:"w-2bfe7e77-8a7c55ab"
id,cur,curStatus,curVal,equipRef,kind,point,unit
@2b80f96a-9dcc2202 "RTU-1 ReturnTemp",M,"ok",68.2°F,@2b80f96a-ee1d1270 "RTU-1","Number",M,"°F"
@2b80f96a-7b16eb30 "RTU-1 ZoneTemp",M,"ok",74.1°F,@2b80f96a-ee1d1270 "RTU-1","Number",M,"°F"
```

You'd be able to iterate over the rows to pick out the value and status of each point:

```python
for row in watchpol_res:
    print ("point %s (ID %s) has value %s (status %s)" \
        % (row["id"].value, row["id"].name, row["curVal"], row["curStatus"]))
```